Z-Bolt/OctoScreen #2: no screens found

negativ72rus, 2019-06-29:
octoprint-tft.service падает с ошибкой:
`octoprint-tft.service - OctoPrint-TFT
   Loaded: loaded (/lib/systemd/system/octoprint-tft.service; enabled; vendor preset: enabled)
   Active: activating (start) since Sat 2019-06-29 10:44:15 BST; 6s ago
 Main PID: 992 (xinit)
   CGroup: /system.slice/octoprint-tft.service
           └─992 /usr/bin/xinit /usr/bin/OctoPrint-TFT -- :0 -nolisten tcp -nocursor

Jun 29 10:44:15 octopi xinit[992]: (EE)
Jun 29 10:44:15 octopi xinit[992]: Fatal server error:
Jun 29 10:44:15 octopi xinit[992]: (EE) no screens found(EE)
Jun 29 10:44:15 octopi xinit[992]: (EE)
Jun 29 10:44:15 octopi xinit[992]: Please consult the The X.Org Foundation support
Jun 29 10:44:15 octopi xinit[992]:          at http://wiki.x.org
Jun 29 10:44:15 octopi xinit[992]:  for help.
Jun 29 10:44:15 octopi xinit[992]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
Jun 29 10:44:15 octopi xinit[992]: (EE)
Jun 29 10:44:15 octopi xinit[992]: (EE) Server terminated with error (1). Closing log file.`

alexkv, 2019-06-29:
Да, вы используете старую версию, сегодня завтра я выложу актуальную. 
Кстати какое разрешение вы используете? (это не связано с проблемой, исключительно для общего развития)

negativ72rus, 2019-06-29:
@alexkv пока что самый дешевый экран из наличия в офлайне https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A) 480x320
Но если  зайдет, закажу что нибудь с HDMI интерфейсом и IPS.
Так же интересно какой экран выберите Вы, планируется ли разработка крепления к болту или самому моделить? :)

alexkv, 2019-07-01:
Актуальная версия: https://github.com/Z-Bolt/OctoScreen/releases
Ошибка должна починиться установкой драйвера: https://github.com/ssvb/xf86-video-fbturbo/wiki/Installation
Эркан адаптирован под разрешение 800x480, поэтому если у вас физическое разрешение экрана 480x320, рекоммендую указать в /boot/config.txt указать разрешение 800х533 (hdmi_cvt 800 533 60 6 0 0 0)

negativ72rus, 2019-07-02:
@alexkv спасибо! обязательно попробую. Тут правда есть нюанс, эран у меня по spi подключается а не по hdmi. Это я про конфигурацию приложения. Что мне прописать в конфиг тогда?

alexkv, 2019-07-02:
Значения теже (800х533), а вот как именно это в конфиге под spi выглядит не
подскажу, не приходилось сталкиваться с такими экранами.

negativ72rus, 2019-07-02:
Ок, буду пробовать :)

negativ72rus, 2019-07-04:
@alexkv все получилось, спасибо!
один момент заработало только так:
OCTOSCREEN_RESOLUTION=480x320

closed 2019-07-04 by negativ72rus