Z-Bolt/OctoScreen #10: Memory leak?

steviehs, 2019-08-05:
From what I see, there are some open issues in the original project, which should now be resolved here. One is a memory leak (hey, wasn't that the reason for golang?): If I have octoscreen running and the printer is disconnected, I see the screen "waiting for connection" - which has do be discussed in another issue - still after some hours, even octoprint get's killed as octoscreen is eating up more and more memory on my RPI3.

alexkv, 2019-08-23:
Does it happen only when OctoPrint is disconnected? How fast does octoscreen eat memory on your environment?

alexkv, 2019-08-23:
Could you check this behaviour with a new version https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.2.0?

steviehs, 2019-08-23:
I don't want to run the printer over night just connected. So it happened without the printer connected in about 8 hrs over night on a rpi3b. I will try the new version when I am back home.

steviehs, 2019-09-09:
Ok, checked it now - and it is the same behaviour. I guess you can easily reproduce it: just let octoscreen and octoprint run for 1-2 days without having the printer on/connected. After that ocotoscreen has eaten up so much memory, that a rpi3b+ won't have enough mem for octoprint anymore...

```
 PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                       
18549 root      20   0 1056732  56672  43064 S   2.0  6.3  29:07.21 OctoScreen        
```

MatthewCroughan, 2019-11-21:
@steviehs this has to be an issue with Octoprint itself, no? It's the serial requests to it that make Tornado run out of memory. Octoscreen is causing the memory leak in Octoprint.

MatthewCroughan, 2019-11-21:
also, do you think you could link those issues in Octoprint you were referring to?

steviehs, 2019-11-21:
Checking over night, Octoprint does not use more memory, but OctoScreen does. So I would see this as an issue of OctoScreen. 
Octoprint is running for months with the printer shut down, but OctoScreen dies approx after 15 hr or so on a rpi3b+

MatthewCroughan, 2019-11-21:
@steviehs For me, as I've linked in the issue, it's clearly Octoprint that's causing a memory leak. Have you got logs? Take a look at mine in https://github.com/Z-Bolt/OctoScreen/issues/58

steviehs, 2019-11-21:
As said, it is not Octoprint growing in mem usage but Octoscreen. So maybe you have another issue. I am from my side not really looking into Octoscreen at this point, as long as I do not see any improvements to this bug. I do not have time nor understanding to trace this bug in go, so unless anybody else is looking at it I have to wait.

MatthewCroughan, 2019-11-21:
I think I understand what you're saying. You believe that Octoscreen grows in memory usage, to the point at which Octoprint is unable to allocate memory? The problem is that when I do `free -m` I don't see a reason for this, both applications should be able to live, there's always 500M free.

MatthewCroughan, 2019-11-21:
@steviehs Issue 56 explains why I believe that this is the fault of Octoprint. Because restarting Octoprint fixes the issue. The state of Octoscreen does not matter.

steviehs, 2019-11-21:
@MatthewCroughan so this makes me pretty sure, this is not connected to the bug I reported: it takes around 5 hr, when I kill OctoScreen then, everything is fine again. Maybe you should leave your bug in your report and let this bug still have it's own little life :-)

MatthewCroughan, 2019-11-21:
Well they're very similar, and probably caused by the same code. So I think they're related still. We can't know whether Octoprint has a leak of its own until we solve this one.

alexkv, 2019-12-25:
Try a new version https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.5.1

closed 2019-12-25 by alexkv

MatthewCroughan, 2019-12-25:
@alexkv Merry christmas, thanks so much. This is a hard problem, because the issue exists in Octoprint in reality. I work around it by doing a similar thing to what you're doing, https://github.com/MatthewCroughan/octobalena/commit/719f29c2f99707c2dabcb025b522057d040eb35a

alexkv, 2019-12-25:
Merry Christmas @MatthewCroughan. You are welcome.

MatthewCroughan, 2019-12-27:
@alexkv This fix is more of a workaround, as the amount of threads still accumulate in Octoprint.
Do you think it would be possible to remove the functionality entirely and stop Octoscreen from trying to connect to Octoprint?

I've attempted this, but whatever I did broke the ui.update function in a way that causes it to crash Octoscreen after a print has been going for a while.

MatthewCroughan, 2019-12-27:
For example, this is the view from htop after just one attempt by octoscreen to reconnect multiple times.

[image]

Octoprint has spawned 33 threads due to 33~ connection attempts by Octoscreen, the only workaround to this that makes sense is the one that I have produced via MatthewCroughan/octobalena@719f29c where there is a container on the device listening for USB serial devices, when they are plugged in it issues a curl command to connect Octoprint to the printer. 

Sadly, connection attempts via the api causing a memory leak is not considered a problem by the maintainer of Octoprint, so this is not likely to be solved upstream any time soon, and I do not have the expertise to debug the web framework used by Octoprint which has a sort of history of these problems when people try to produce applications with it.

https://github.com/foosel/OctoPrint/issues/3376
https://github.com/foosel/OctoPrint/issues/2585

This is in fact why Octoprint is stuck on a very old version of Tornado.

MatthewCroughan, 2019-12-27:
This would be an acceptable solution, and much cleaner if it didn't continue to poll the printer when it didn't connects, that way the memory leak can only continue to occur *if* the user clicks retry whilst the printer is disconnected. 

The problem now is that on 2.5.1 the memory leak is going to get worse every single time the printer disconnects, the problem still exists it has just been moved around.

thebeline, 2021-03-16:
Hey all: This polling issue has a fix in 2.7.0-dev.  By default it will now only poll for a printer connection 8 times over 80 seconds (once every 10 seconds) and then go into a hold state.  The user then has the option to re-start the polling process.  This should mitigate this issue.

There is also an additional PR in progress with regards to the auto-connect nature of OctoScreen, #252 

This PR allows the user to disable the auto-connect feature, instead relying on OctoPrint to handle connecting.  However, it also gives the user the option (after 30 seconds) to manually trigger an auto-connect.

Feedback on `2.7.0-dev` and #252 would be fantastic.