Z-Bolt/OctoScreen #12: v2.2: Layout broken after entering the System screen

gdachs, 2019-08-24:
Hello,
in the System screen the back button is shifted out of the display. It is hard to reach. If you return from the screen all other screens have the same layout problem. If you go to the Print screen you will never get out again. My resolution is 800x480. In version 2.1 there is no problem.
[image]
[image]

IdefixRC, 2019-08-25:
did you add custom controls? I found that if I add custom controls with long names I run into the same issue after exiting the control screen.

gdachs, 2019-08-25:
No, I don't use custom controls. I have now disabled all non default plugins with no effect. This started with v2.2. I have no problems with v2.1.

alexkv, 2019-08-30:
Fixed this issue: https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.2.1

closed 2019-08-30 by alexkv