Z-Bolt/OctoScreen #13: V2.1: Custom Controls with long names causing layout issue

IdefixRC, 2019-08-25:
Hi,

Running Octoscreen V2.1 on OctoPrint 1.3.11 running on OctoPi 0.16.0.
Resolution set to 800x533.

I have played around with Octoscreen and ran into an issue with custom controls (added through Custom Control Editor Plugin). After adding all my custom controls the control screen still showed the back button but upon return to the main menu the screen was stretched and the back buttons of all other menus no longer accessible.
After that, I removed a number of controls and see if it improves but the issue remained or the back button in the control screen disappeared altogether. 
After removing all controls it worked fine again.

Next, I added all the controls back and checked after each section.
As soon as I added buttons with longer names such as "Load Filament (Sensor)" the error occurred again. Reducing the text to "Load (Sensor)" fixed the issue.

Checked the max length after that and it looks like up to 16 char work fine, after that the layout gets shifted out of the frame and runs into issues.

Hope this helps to troubleshoot for the next release.

gdachs, 2019-08-25:
sorry for the unrelated question, but did you edit the styles.css? My background looks grey, yours blue.

IdefixRC, 2019-08-26:
No, I did not change anything. Could simply be due to the angle or ambient that makes it look blueish. Believe (need to check after work) that it is black :-)

alexkv, 2019-09-23:
@IdefixRC, try this screen in new version https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.3.0

closed 2019-10-16 by alexkv