Z-Bolt/OctoScreen #14: V2.1 Feature Request - button code to to be stored in external file

IdefixRC, 2019-08-25:
Hi,

if I read the code correctly then commands such as filament loading or unloading from the filament screen are hard-coded. As there are many printers out there which special commands (e.g. the filament load commands currently implemented are running my extruder on overdrive and will grind the filament down before it can even be loaded), is it possible to instead store the scripts in external files so that they can - if necessary - be changed or replaced with printer specific code?

Octoprint uses the ~/.octoprint/scripts/gcode folder for custom command scripts. Adding them here would consolidate them all in the same place (e.g. call them oscreen_load.gco etc..)

Thanks !

kizza42, 2019-09-10:
This is a good suggestion, I accidentally pressed "Unload Filament" today with a cold hotend, lots of grinding ensued. I'd like to edit the buttons to pre warm the hot end etc.
Similarly the Fast/Slow moves are too fast/slow for my printer.

alexkv, 2019-11-23:
Implemented in the new version https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.4.9

closed 2019-11-23 by alexkv