Z-Bolt/OctoScreen #17: Installation

MasterTim17, 2019-09-06:
Hello,
i would like to have this OctoScreen running on my raspberry pi 3b+. I have this display [TFT 3.5](https://www.banggood.com/3_5-Inch-320-X-480-TFT-LCD-Display-Touch-Board-For-Raspberry-Pi-2-Model-B-RPI-B-p-1023432.html?p=R90804949020201412E4&cur_warehouse=CN) and followed Step 3&4 from this page [TFT Drivers](https://www.elecrow.com/wiki/index.php?title=3.5_Inch_480x320_TFT_Display_with_Touch_Screen_for_Raspberry_Pi). On reboot i see normal start of raspberry.

Following the instructions of this repository i executed
`sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils`
and
`wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.2.1/octoscreen_2.2-2_armhf.deb`
`sudo dpkg -i octoscreen_2.2-2_armhf.deb`
and also activated auto console login in raspi-config.
But after reboot only console pi@octopi:~$ with cursor blinking shows up.

How do i get OctoScreen to run?

rocky00717, 2019-09-07:
also having the same problem

MonDroone, 2019-09-11:
same problem for me

MonDroone, 2019-09-11:
I followed these instructions.
Octoscreen started well but, after rebooting the raspberry, octoscreen does not restart itself.
Is there a solution?
##################################################
How to update
Run the following commands:
wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.2.1/octoscreen_2.2-2_armhf.deb
sudo dpkg -r octoscreen
sudo dpkg -i octoscreen_2.2-2_armhf.deb
sudo systemctl restart octoscreen.service
#################################################

For start Octoscreen, it is necessary to execute this code via putty
sudo systemctl restart octoscreen.service

rocky00717, 2019-09-12:
I had the wrong driver installed.  Once I installed the waveshare driver and reinstalled octoscreen, it started working.

alexkv, 2019-09-15:
Did you install [xf86-video-fbturbo](https://github.com/ssvb/xf86-video-fbturbo/wiki/Installation)? It's mandatory for some types of screens.

fsalinasc, 2019-09-22:
I have the same problem.
Any way to auto start?

alexkv, 2019-09-22:
Run command:
`sudo service octoscreen status`
And show what it returns

MonDroone, 2019-09-22:
After >>> sudo service octoscreen status
I have this return :
● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor prese
   Active: inactive (dead)

lines 1-3/3 (END)

alexkv, 2019-09-23:
Did you try to reinstall octoscreen once again?
What hardware and OS do you use?

MasterTim17, 2019-09-23:
im using latest octoprint image. 
this is the result of _sudo service octoscreen status_
`● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor prese
   Active: activating (start) since Mon 2019-09-23 18:12:35 BST; 1s ago
 Main PID: 1900 (xinit)
   CGroup: /system.slice/octoscreen.service
           ├─1900 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocurs
           ├─1902 /usr/lib/xorg/Xorg :0 -nolisten tcp -nocursor
           └─1919 /usr/bin/OctoScreen`

fsalinasc, 2019-09-24:
> Did you try to reinstall octoscreen once again?
> What hardware and OS do you use?

I have reinstalled octoscreen and it remains the same.
Raspberry Pi 3 and Raspbian duster.
To make it work, every time I restart, I have to send by ssh
sudo systemctl restart octoscreen.service

fsalinasc, 2019-09-24:
> Did you try to reinstall octoscreen once again?
> What hardware and OS do you use?

I have reinstalled octoscreen and it remains the same.
Raspberry Pi 3 and Raspbian Stretch.
To make it work, every time I restart, I have to send by ssh
sudo systemctl restart octoscreen.service

fsalinasc, 2019-09-24:
> > Did you try to reinstall octoscreen once again?
> > What hardware and OS do you use?
> 
After each reboot, I have to send by ssh
sudo systemctl restart octoscreen.service
When octoscreen is already running, shipping
sudo service octoscreen status
and this is the result:

● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-09-24 20:55:02 BST; 19s ago
  Process: 1013 ExecStartPost=/bin/bash /etc/octoscreen/disablescreenblank.sh 0 (code=exited, status=0/SUCCESS)
 Main PID: 990 (xinit)
   CGroup: /system.slice/octoscreen.service
           ├─ 990 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor
           ├─ 992 /usr/lib/xorg/Xorg :0 -nolisten tcp -nocursor
           └─1001 /usr/bin/OctoScreen

Sep 24 20:54:51 octopi xinit[990]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Sep 24 20:54:51 octopi xinit[990]: (==) Log file: "/var/log/Xorg.0.log", Time: Tue Sep 24 20:54:51 2019
Sep 24 20:54:51 octopi xinit[990]: (==) Using config file: "/etc/X11/xorg.conf"
Sep 24 20:54:51 octopi xinit[990]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Sep 24 20:54:51 octopi xinit[990]: modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/4.14.79-v7+
Sep 24 20:54:52 octopi xinit[990]: time="2019-09-24T20:54:52+01:00" level=info msg="OctoPrint's config file found: \"/home/pi/.octoprint/config.yaml\"" file=main.go func=main.readConfig line=1
Sep 24 20:54:52 octopi xinit[990]: time="2019-09-24T20:54:52+01:00" level=info msg="Using \"http://localhost:5000\" as server address" file=main.go func=main.init.0 line=52
Sep 24 20:54:52 octopi xinit[990]: time="2019-09-24T20:54:52+01:00" level=info msg="Found API key at \"/home/pi/.octoprint/config.yaml\" file" file=main.go func=main.init.0 line=58
Sep 24 20:54:52 octopi xinit[990]: time="2019-09-24T20:54:52+01:00" level=debug msg="New background task started" file=common.go func="ui.(*BackgroundTask).Start" line=114
Sep 24 20:55:02 octopi systemd[1]: Started OctoScreen.

lele344, 2019-10-29:
I've the same issue. The octoscreen service doesn't start on boot.
I've fixed it adding the "sudo systemctl start octoscreen.service" line into the /etc/rc.local for now, and in this way, I can have octoscreen on boot, without send ssh command. 
It's only a workaround but works fine for me for now.

nettunoroberto, 2019-12-01:
sudo dpkg -i octoscreen_2.2-2_armhf.deb
Selecting previously unselected package octoscreen.
(Reading database ... 73940 files and directories currently installed.)
Preparing to unpack octoscreen_2.2-2_armhf.deb ...
Unpacking octoscreen (2.2-2) ...
dpkg: dependency problems prevent configuration of octoscreen:
 octoscreen depends on libgtk-3-0 (>= 3.21.5); however:
  Version of libgtk-3-0:armhf on system is 3.14.5-1+deb8u1rpi1rpi1g.

dpkg: error processing package octoscreen (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 octoscreen

shilbert01, 2020-01-01:
Check issue #34. The service needs to be enabled. It might come up as active but "dead". Check with systemctl. If so, make sure you have set the Pi to boot into graphical environment with login (might work without auto-login).

tiagomaricate, 2020-08-15:
Does anybody manage to fix this error?

```
dpkg: dependency problems prevent configuration of octoscreen:
 octoscreen depends on libgtk-3-0 (>= 3.21.5); however:
  Version of libgtk-3-0:armhf on system is 3.14.5-1+deb8u1rpi1rpi1g.
```

JeffB42, 2021-01-05:
@MasterTim17 there are some set up instructions posted at
https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and
https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

Let me know if this helps and if you are up and running.

MasterTim17, 2021-01-05:
> @MasterTim17 there are some set up instructions posted at
> https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and
> https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen
> 
> Let me know if this helps and if you are up and running.

I found this instruction, but the struggling with the spi display was too much, so I bought a hdmi display
Thanks for the help

closed 2021-01-05 by MasterTim17