Z-Bolt/OctoScreen #21: V2.3.0: screens are shifted to the right after entering the control screen

gdachs, 2019-09-24:
Just installed version 2.3.0 on Ubuntu on my Odroid C2. Looks fine so far, good job!

Small problem is that after entering the control screen, every screen is shifted slightly to the right, so a small part gets cuttet. Not so much like the other bug I posted some time ago with the version 2.1.

My resolution is 800x480.

gdachs, 2019-10-04:
in 2.3.0-2 still not fixed

alexkv, 2019-10-05:
Please attache, photo of your control screen.

gdachs, 2019-10-05:
Before entering the Control screen, everything is okay:
[image]
Inside the Control screen, text to the right is cuttet:
[image]
Are there some SVG files missing?
After leaving the Control screen, the screen is shifted to the right, this will happen with all screens:
[image]

alexkv, 2019-10-15:
Try now: https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.3.5

gdachs, 2019-10-15:
perfect!

closed 2019-10-15 by gdachs