Z-Bolt/OctoScreen #24: New install, screen not lining up

Vortec4800, 2019-09-27:
Just installed on R pi 3b+ with an Adafruit 3.5" touchscreen. Did the Adafruit driver install and touchscreen calibration, and their touchscreen test tool shows everything is accurate.

However when I launch OctoScreen, everything appears to be off to the right, and the touch input seems backwards. If I touch on the left side of the screen, buttons on the right side activate, etc. Attached photos for reference.

[image]
[image]
[image]

FlyingT, 2019-10-06:
Same here, any help?
I set OCTOSCREEN_RESOLUTION to 800x533 as stated for 480x320 displays ..

acorderob, 2019-10-12:
Did you also change the hdmi_cvt entry in /boot/config.txt file?

FlyingT, 2019-10-12:
> Did you also change the hdmi_cvt entry in /boot/config.txt file?

Why? Its a GPIO Display, not a HDMI

acorderob, 2019-10-13:
I don't know, but the display is set to mirror the hdmi, so that might affect it.

JeffB42, 2021-01-06:
@Vortec4800 Is this still an issue?  Did you get your display working?  Here are some set up guides:
https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display
and https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

closed 2021-03-27 by JeffB42