Z-Bolt/OctoScreen #25: Feature Request: Launch Cura via EGL or some other method for slicing. Similar functionality to astroprint

MatthewCroughan, 2019-09-27:
This project could be way ahead in its simplicity if it were possible to launch Cura on something like a Pi4 via a menu item with a hotkey to quit via either a hotkey on GPIO or on-screen overlay. Having some slicing functionality available without the commercial plague that has afflicted the Astroprint stuff right there on the display would be great for a makerspace I attend.

I'm putting this here because it's something I want badly, so I'll also be making some efforts to try and implement this once I familiarize myself with the codebase here. I'm so glad you started this.

[image]

alexkv, 2019-09-28:
Sound great! We can user slicing features that OctoPrint already has. The only lack that I see here is an old version of Cura that OctoPrint supported. Also, pay attention that OctoScreen UI is based on GTK3.

MatthewCroughan, 2019-09-28:
@alexkv The reason I want to use Cura is because it looks the part. It already has Octoprint integration via a plugin, and it is more intuitive for new users especially in a public space. Why does GTK matter if we're launching