Z-Bolt/OctoScreen #27: unable to install screen flashes waveshare driver

mvoss96, 2019-09-29:
Hi, I'm unable to install octoscreen when the waveshare driver is installed (https://github.com/waveshare/LCD-show) 

I'm using the follwing driver: ./LCD35-HDMI-480x320-show

i tried it with the current octopi image /stretch) as well as with a stock debian buster in both cases the touchscreen works, but after installing octoscreen with the .deb i get the following error:
Failed to preset unit: File /etc/systemd/system/display-manager.service already exists and is a symlink to /lib/systemd/system/lightdm.service.
/usr/bin/deb-systemd-helper: error: systemctl preset failed on octoscreen.service: No such file or directory 

after reboot the raspberry just shows the console and flashes black every few seconds#

on debian buster (with desktop) i can still ipen the desktop with startx, but octoscreen doesnt start

italocjs, 2019-10-31:
On this one:   File /etc/systemd/system/display-manager.service already exists and is a symlink to /lib/systemd/system/lightdm.service.

After installing the lightdm i choose it to run the octoscreen, so i dont get this error,   (actually i had to install a bunch of stuff, it would be really easier if there was a prepacked image),  Now i can see the image of the terminal on the screen, but octoscreen still saying no screens found

JeffB42, 2021-03-27:
@mvoss96 is this still an issue?  Please look over https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen.  Also, v2.7.0 was just released.  I suggest installing the latest as it fixes a lot of bugs.

closed 2021-04-17 by JeffB42