Z-Bolt/OctoScreen #28: Octoscreen not started

labomatik, 2019-09-30:
I'm not able to have octoscreen running, when i'm checking the status i've this:
`pi@octopi:~ $ systemctl status octoscreen.service
● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: activating (start) since Mon 2019-09-30 11:08:22 BST; 6s ago
 Main PID: 4009 (xinit)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/octoscreen.service
           └─4009 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor

Sep 30 11:08:22 octopi xinit[4009]: (EE)
Sep 30 11:08:22 octopi xinit[4009]: Fatal server error:
Sep 30 11:08:22 octopi xinit[4009]: (EE) Server is already active for display 0
Sep 30 11:08:22 octopi xinit[4009]:         If this server is no longer running, remove /tmp/.X0-lock
Sep 30 11:08:22 octopi xinit[4009]:         and start again.
Sep 30 11:08:22 octopi xinit[4009]: (EE)
Sep 30 11:08:22 octopi xinit[4009]: Please consult the The X.Org Foundation support
Sep 30 11:08:22 octopi xinit[4009]:          at http://wiki.x.org
Sep 30 11:08:22 octopi xinit[4009]:  for help.
Sep 30 11:08:22 octopi xinit[4009]: (EE)`

The screen on the pi is displaying the desktop.

Any idea?

gdachs, 2019-09-30:
Did you try what the error message suggests?

labomatik, 2019-09-30:
sure, after the delete of /tmp/.X0-lock

`pi@octopi:~ $ systemctl status octoscreen.service
● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: activating (start) since Mon 2019-09-30 11:18:32 BST; 5s ago
 Main PID: 4323 (xinit)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/octoscreen.service
           └─4323 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor

Sep 30 11:18:32 octopi xinit[4323]: (EE)
Sep 30 11:18:32 octopi xinit[4323]: Fatal server error:
Sep 30 11:18:32 octopi xinit[4323]: (EE) Cannot establish any listening sockets - Make sure an X server isn't already running(EE)
Sep 30 11:18:32 octopi xinit[4323]: (EE)
Sep 30 11:18:32 octopi xinit[4323]: Please consult the The X.Org Foundation support
Sep 30 11:18:32 octopi xinit[4323]:          at http://wiki.x.org
Sep 30 11:18:32 octopi xinit[4323]:  for help.
Sep 30 11:18:32 octopi xinit[4323]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
Sep 30 11:18:32 octopi xinit[4323]: (EE)
Sep 30 11:18:32 octopi xinit[4323]: (EE) Server terminated with error (1). Closing log file.`

gdachs, 2019-09-30:
And you checked on the system that it is really not running?

labomatik, 2019-10-01:
I've no idea how to check that. I'm not familiar with Xserver

gdachs, 2019-10-01:
It is not specific to Xserver to check whether it is running. 
You could use `ps -ef | grep X`

labomatik, 2019-10-02:
`pi@octopi:~ $ ps -ef | grep X
root      7339  4374  0 07:10 tty7     00:00:00 /usr/lib/xorg/Xorg :0 -seat seat0 -auth /var/run/lightdm/root/:0 -nolisten tcp vt7 -novtswitch
root      7452  7451  0 07:11 ?        00:00:00 [Xorg] <defunct>
root      7455  7454  0 07:11 ?        00:00:00 [Xorg] <defunct>
pi        7465  7271  0 07:11 pts/0    00:00:00 grep --color=auto X`

gdachs, 2019-10-02:
as you can see it is running already.

tubbers, 2019-11-24:
i am having some issues too. I try to set octoscreen with sudo dpkg-reconfigure octoscreen but it complains about lightDM being missing

pi@octopi:~ $ systemctl status octoscreen.service
● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: activating (start) since Sun 2019-11-24 11:48:06 AEDT; 728ms ago
 Main PID: 2450 (xinit)
   CGroup: /system.slice/octoscreen.service
           ├─2450 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor
           └─2451 /usr/lib/xorg/Xorg :0 -nolisten tcp -nocursor

Nov 24 11:48:06 octopi xinit[2450]: xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support)
Nov 24 11:48:06 octopi xinit[2450]: Current version of pixman: 0.34.0
Nov 24 11:48:06 octopi xinit[2450]:         Before reporting problems, check http://wiki.x.org
Nov 24 11:48:06 octopi xinit[2450]:         to make sure that you have the latest version.
Nov 24 11:48:06 octopi xinit[2450]: Markers: (--) probed, (**) from config file, (==) default setting,
Nov 24 11:48:06 octopi xinit[2450]:         (++) from command line, (!!) notice, (II) informational,
Nov 24 11:48:06 octopi xinit[2450]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Nov 24 11:48:06 octopi xinit[2450]: (==) Log file: "/var/log/Xorg.0.log", Time: Sun Nov 24 11:48:06 2019
Nov 24 11:48:06 octopi xinit[2450]: (==) Using config directory: "/etc/X11/xorg.conf.d"
Nov 24 11:48:06 octopi xinit[2450]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
pi@octopi:~ $ sudo dpkg-reconfigure octoscreen
Installing service /lib/systemd/system/lightdm.service
WARNING: /lib/systemd/system/lightdm.service is the selected default display manager but does not exist

shilbert01, 2020-01-01:
dpkg-reconfigure seems to set the displaymanager to lightdm no matter what you choose in the dialogue. However
`sudo systemctl disable lightdm`

gave this

`Synchronizing state of lightdm.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable lightdm
Removed /etc/systemd/system/display-manager.service.`

Then, 
`sudo systemctl enable octoscreen.service`

give this
 
`Created symlink /etc/systemd/system/display-manager.service → /lib/systemd/system/octoscreen.service.
Created symlink /etc/systemd/system/graphical.target.wants/octoscreen.service → /lib/systemd/system/octoscreen.service.`

After this, do not use dpkg-reconfigure anymore for octoscreen because (I believe) it will once again reset this to lightdm

I had to set the PI to "autologin into Desktop" via `sudo raspi-config`

A reboot brought the system up and displayed the OctoScreen UI

JeffB42, 2021-03-27:
@labomatik is this still an issue? Please look over https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen. Also, v2.7.0 was just released. I suggest installing the latest as it fixes a lot of bugs.

closed 2021-04-17 by JeffB42