Z-Bolt/OctoScreen #30: Octoscreen service not starting on boot

drphil3d, 2019-10-02:
I'm not sure why it's not been registered as a service to start on boot. any ideas?

shilbert01, 2020-01-01:
It is registered if you use `sudo systemctl enable octoscreen.service`

It will give you this output.
 
`Created symlink /etc/systemd/system/display-manager.service → /lib/systemd/system/octoscreen.service.
Created symlink /etc/systemd/system/graphical.target.wants/octoscreen.service → /lib/systemd/system/octoscreen.service.`

If it still does not come up check if it is running or not with

`sudo systemctl status octoscreen.service`

If it says active but "dead" this means it has been loaded but it not displayed. In my case I had to use raspi-config to boot into graphical environment.

You might want to have a look at issue #34 as well

JeffB42, 2021-02-04:
@drphil3d is this still an issue?

JeffB42, 2021-03-27:
Closed due to lack of response.

closed 2021-03-27 by JeffB42

ivizot, 2021-04-26:
Hey all,

I think the problem with "dead" service is related to the following line in  OctoScreen/debian/octoscreen.service:
`WantedBy=graphical.target`

In modern OctoPi images, there is no way (by default) to enable GUI autologin through `raspi-config` without installing of a display-manager.

Thus, the workaround might be to change target status required by OctoScreen?

nathan-stewart, 2023-09-26:
Just wanted to underscore ivizot's comment as the key in 2023.

FredFoo, 2024-03-03:
Ah, nice!

Had a look into the raspi-config script, interested in what it is doing.  So if you want to get this to work without installing another display manager, do this:

1. In raspi-config activate "Console Autologin" instead of desktop. This activates autologin
2. On the commandline, issue 'systemctl set-default graphical.target', to change the target manually

Thats it. 

BTW, you can manually activate autologin by creating the file /etc/systemd/system/getty@tty1.service.d/autologin.conf and add these lines:
`
[Service]
ExecStart=
ExecStart=-/sbin/agetty --autologin pi --noclear %I $TERM
`
Is what the config script does.

zapdogster, 2024-11-22:
Hey @FredFoo  - your solution worked perfectly for me. I tried all sorts of other "stuff" from other threads on here, and none of them worked. Yours did. THANKS!