Z-Bolt/OctoScreen #31: New Feature: PSU Control

MrFlexon, 2019-10-06:
It would be great to shutdown and start the printer from the raspberry. Some plugin is already available in octoprint (by Shawn Bruce: psu control).

ChaosBlades, 2020-03-15:
Well, I guess if you use PSU Control you can't use OctoScreen without some major inconvenience. If you have PSU Control setup you have to open OctoPrint from your phone or webui, turn on the printer and then connect to your printers main board via USB serial connection before you can get past the OctoScreen "Initializing" screen defeating the purpose of having a touchscreen in the first place.

I think using PSU Control with a relay is a fairly common mod or maybe not with basically no comments here. However, it would be nice to get this out of backlog.

michielbaird, 2020-03-15:
Checkout my fork

Rshep67, 2020-03-16:
If you use a power strip and tie your PSU into that, you can turn off multiple equipment. Pi, Printer....etc just make sure you use the web II to shut down the Pi first. Relay should be robust enough to handle the amps so watch you numbers.

Regards, Rob Shepherd

ChaosBlades, 2020-03-16:
@michielbaird Do you have a pre-compiled deb package for OctoPi? I am not familiar with compiling from source on linux.

@Rshep67 I use a normally open 30A relay and use the Temperature Failsafe Octoprint plugin as a secondary form of thermal runaway.

michielbaird, 2020-03-17:
https://github.com/michielbaird/OctoScreen/releases/tag/2.5.1-psu-control

ChaosBlades, 2020-03-18:
Thanks @michielbaird you are a legend. But how is your version different? I still need to go into the WebUI to press the power button before it will connect. Just like before I also get a constant stream of "Error connecting to printer" messages when the printer is off.

Is it looking for a connection before it powers on the printer? If that is the case I have an EZBoard Lite. The USB port on that main board does not accept USB power to power on the board. Because you shouldn't be trying to power the board with USB power anyways because it also tries to power the entire printer including stepper motors and the heaters through the 5V on the pi.

It really needs a power button on the initialize screen.

ChaosBlades, 2020-03-18:
@michielbaird I think you may have built the master branch and not the 'psucontrol-integration' branch? It mentions a button you added and I think I have been through all the menus and didn't find any button to control the PSU.

michielbaird, 2020-03-18:
So sorry will get you the right version asap

michielbaird, 2020-03-18:
Yeah I grabbed the old binary... It's been updated now.

ChaosBlades, 2020-03-18:
Excellent! It works as it should.

paulsijben, 2020-05-17:
alas it does not quite work for me. It does show the PSU state but it will no longer accept any input from the touch screen. After reinstalling the regular release, it works again. 

Anything I can do to help debut this? As I am looking for this feature!

ChaosBlades, 2020-05-17:
I think this project is dead. It also has an issue where it restarts the service every so many minutes that no one has been able to fix. Use OctoDash instead.

https://github.com/UnchartedBull/OctoDash

clough42, 2020-05-17:
Another option would be to keep the 5V power on and only use the PSU switch to shut down the 12V rail.  I run my raspberry pi and the 5V printer circuitry from the 5VSB rail of the supply, which is always on, and use the PSU switch to turn the 12V supply for the motors and heaters on and off.  This way, everything stays connected and you can use M80/M81 to turn the heater/motor supply on and off.

This of course depends on your hardware.

floridaservices, 2020-05-17:
> I think this project is dead. It also has an issue where it restarts the service every so many minutes that no one has been able to fix. Use OctoDash instead.
> 
> https://github.com/UnchartedBull/OctoDash

Octodash has a very active and responsive developer right now, I recommend it also

ivizot, 2021-04-08:
Dear @alexkv Alexey, could you please look into integration of PSU in your awesome Octoscreen? It is really a must-have feature...

JeffB42, 2021-04-09:
I don't think Alexey works on OctoScreen anymore.  I've been working on it and implementing new features.  Adding a PSU control is on my radar, though I'm not sure when it will be in.  What complicates things is that it seems like a button which controls the PSU would interact with the PSU plug-in (https://github.com/kantlivelong/OctoPrint-PSUControl), but would need to be on the splash screen, which is what you see if the printer isn't on, but is also what you see if OctoScreen isn't able to connect to OctoPrint.  In 2.8 I'm going to be looking at the state machine, and I might look into implementing this then.

ivizot, 2021-04-09:
> I don't think Alexey works on OctoScreen anymore. I've been working on it and implementing new features. Adding a PSU control is on my radar, though I'm not sure when it will be in. What complicates [...]

Great! Other option might be to not directly interact with PSU, but rather simply toggle RPi GPIO pin (if PSU is physically controlled by RPi, which is my use-case :) )
Also, I have found several issues with OctoScreen when working with multiple-head machine. If it is of interest to you, should I open a new issue?

JeffB42, 2021-04-09:
for the other issues, open up a new ticket.

kantlivelong, 2021-04-16:
@JeffB42 If you do implement this keep in mind that not all configurations have the controller powered on/off via the plugin. Setups such as my own are merely to toggle 12V for motors, hotends, fans, and other devices that required 12V but don't need to be on all the time.