Z-Bolt/OctoScreen #33: RPi3B / 3,5" GPIO Screen - unable to make it fit

FlyingT, 2019-10-10:
Whatever I do, its impossible to make the screen scale to the correct size.

mdaneman, 2019-10-13:
I see this issue unanswered in a few posts.  Does that mean that Octoscreen doesn't work for GPIO 320x480 screens?  Or is there a solution?

chilman408, 2019-10-15:
Did you guys try scaling the output screen?

sudo nano /boot/config.txt
# Add this line to the end of the file
 hdmi_cvt 800 533 60 6 0 0 0

Then:
sudo nano /etc/octoscreen/config
# Edit the following line to show this
OCTOSCREEN_RESOLUTION=800x533

mdaneman, 2019-10-15:
Acutally I found the solution here.  hdmi_cvt doesn't work unless you're using HDMI.  The trick, apparently, is to actually use HDMI and use the GPIO LCD screen as a clone by coping the HDMI frame-buffer to it.

[https://github.com/Z-Bolt/OctoScreen/issues/34#issuecomment-541835209](url)

FlyingT, 2019-10-19:
> Acutally I found the solution here. hdmi_cvt doesn't work unless you're using HDMI. The trick, apparently, is to actually use HDMI and use the GPIO LCD screen as a clone by coping the HDMI frame-buffer to it.
> 
> #34
> 
> [https://github.com/Z-Bolt/OctoScreen/issues/34#issuecomment-541835209](url)

So I tried that ..
This is just empty for me:
sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf

any help?

chilman408, 2019-10-20:
> So I tried that ..
> This is just empty for me:
> sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf
>
> any help?

Can you post your steps on trying to install?  Mdaneman is correct, you need to install additional drivers to use HDMI as a clone.

We'll help figure it out.

thenickdude, 2019-12-26:
> ==> Add this line to the end of the file
> hdmi_cvt 800 533 60 6 0 0 0

I also needed to set:

```
framebuffer_width=800
framebuffer_height=533
```

Then it finally worked!

adrianmihalko, 2020-03-05:
> Did you guys try scaling the output screen?
> 
> sudo nano /boot/config.txt
> 
> ==> Add this line to the end of the file
> hdmi_cvt 800 533 60 6 0 0 0
> 
> Then:
> sudo nano /etc/octoscreen/config
> 
> ==>Edit the following line to show this
> OCTOSCREEN_RESOLUTION=800x533
> 
> EDIT: I am using a 3.5" GPIO Screen BTW, all the scaling work fine for me. Or comment back what you'd tried. We'll put our heads together to figure it out.

Thanks, this worked for MPI3508 HDMI display.

JeffB42, 2021-01-06:
@FlyingT and @mcuadros Is this still an issue?  Did you get your display working?  Here are some set up guides:
https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display
and https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

FlyingT, 2021-01-06:
@JeffB42 No, never
Just switched to OctoDash, which scaled perfectly out-of-the-box

closed 2021-01-06 by FlyingT