Z-Bolt/OctoScreen #34: OctoScreen dont start on boot and screen dont fit correctly

ARCH-3D, 2019-10-14:
The OctoScreen seems that could be a fantastic app from what I read :) but it would be great if you could solve the problem that the program it not starting on boot and the screen dont fit correctly. I see that like me others users have the same problem and dont can use an app that could be amazing. Thanks in Advance :)

gdachs, 2019-10-14:
I don't think that it is that easy. I use OctoScreen with Ubuntu installed on an Odroid C2 and it starts just fine. I guess that most users that have problems use Rasbian/OctoPi on an Raspberry Pi. So the problem could be even the operating system and not OctoScreen. Often the Raspberry Pi users are not the most experienced users. If I would have this start problem I would investigate the problem, find the reason and send a fix. 
So don't wait for a fix, just do it yourself and talk about.
First check whether OctoScreen is really not running. Maybe it is running but not visible. If it is not running, try what happens if you start the systemd service by hand ...

ARCH-3D, 2019-10-14:
Thanks for your answer gdachs :) Yes my system is OctoPi on a Raspberry Pi and I also confirm that I am not an experienced user. So I was looking for some help from more experienced users who have had the same problem and solved it. I was looking for solutions before giving up because it looks a very interesting application. I dont know if you could you help me (and others users in the same situation)? :)

gdachs, 2019-10-14:
Oh I think I helped already. Just look whether OctoScreen is running after boot. This is the first step.
The command `ps -ef | grep OctoScreen` should do the job.

ARCH-3D, 2019-10-14:
Thanks gdachs :) Seems that is not running after boot: 
ps -ef | grep OctoScreen
pi        1032   813  0 14:53 pts/0    00:00:00 grep --color=auto OctoScreen
After i do "sudo systemctl restart octoscreen.service" the OctoScreen starts. 

Note: I think that in my case i solved the question of the screen resolution :)

gdachs, 2019-10-14:
Now the question is, did it never start, or it started and stopped afterwords on error. So reboot again and try `sudo systemctl status octoscreen`

ARCH-3D, 2019-10-14:
thanks. 
the result:
octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor prese
   Active: inactive (dead)

gdachs, 2019-10-14:
look like it was never started. At this point the difference between raspbian and ubuntu is too big, I can't really help here. 
What you could try as a hacky workaround is to add the line `systemctl start octoscreen` to the file /etc/rc.local just above the line that contains `exit 0`

ARCH-3D, 2019-10-14:
Thanks a lot gdachs :) 
It works adding "systemctl start octoscreen &" to the file /etc/rc.local. 
Now all seems working :) but I will test ;) Thanks for your help :)

mdaneman, 2019-10-14:
> Thanks gdachs :) Seems that is not running after boot:
> ps -ef | grep OctoScreen
> pi 1032 813 0 14:53 pts/0 00:00:00 grep --color=auto OctoScreen
> After i do "sudo systemctl restart octoscreen.service" [...]

How did you solve the screen resolution issue?  I've tried it as instructed (i.e. set octoscreen resolution to 800x533 and set hdmi_cvt=800 533 60... in boot config) and it still doesn't scale correctly on my 480x320 GPIO LCD.

ARCH-3D, 2019-10-14:
I did it this way:

**Configure Octoscreen:**
sudo nano /etc/octoscreen/config

OCTOSCREEN_RESOLUTION=800x600

**Install fbcp to "clone" the HDMI output:**

sudo apt-get -y install cmake 
git clone https://github.com/tasanakorn/rpi-fbcp 
cd rpi-fbcp
mkdir build 
cd build 
cmake .. 
make 
sudo install fbcp /usr/local/bin/fbcp

**Make fbcp start on every boot:**

sudo nano /etc/rc.local

-> Then, go at the end of file (**but before the exit 0** ) and add a line where to find fbcp :

/usr/local/bin/fbcp &

**Activate and setup HDMI:**

sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf

-> set "fb1 to "fb0

sudo nano /boot/config.txt

change
framebuffer_width=800
framebuffer_height=600

add line
hdmi_cvt 800 600 60 6 0 0 0

Try if it works with you and comment ;)

gdachs, 2019-10-14:
> Thanks a lot gdachs :)
> It works adding "systemctl start octoscreen &" to the file /etc/rc.local.
> Now all seems working :) but I will test ;) Thanks for your help :)

I wonder why you need the "&" ? systemctl is not a background service and finishes after it has started the octoscreen service.

ARCH-3D, 2019-10-14:
> > Thanks a lot gdachs :)
> > It works adding "systemctl start octoscreen &" to the file /etc/rc.local.
> > Now all seems working :) but I will test ;) Thanks for your help :)
> 
> I wonder why you need the "&" ? systemctl is not a background service and finishes after it has started the octoscreen service. Not all commands like if you steal them stdout.

i don't know why but it only works with "&".

FlyingT, 2019-10-19:
This is just empty for me:
sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf

any help?

ARCH-3D, 2019-10-19:
try this:
sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf

Shama71, 2019-10-24:
> This is just empty for me:
> sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf
> 
> any help?
Section "Device"
  Identifier "myfb"
  Driver "fbdev"
  Option "fbdev" "/dev/fb0"
EndSection

xmow49, 2019-11-06:
Hi,
with https://github.com/Z-Bolt/OctoScreen/issues/34#issuecomment-541835209
i have a black screen and sometimes I have a nozzle icon in big on my screen 0.5s and black screen
Can you help me pls
thx

rally25rs, 2020-02-07:
If you get here and are trying to make this work on a Raspberry Pi / Raspien ...

I got mine to work by doing:
```
sudo raspi-config
```
go into "boot" options and set it to boot to a graphical interface with auto-login. Save/exit, choose "no" when it asks if you want to reboot now.

```
sudo systemctl enable octoscreen
```

if it complains that a link is `"already used by xyz.service"` then you probably already had your Pi set to boot into another GUI... do
```
sudo systemctl disable xyz
sudo systemctl enable octoscreen
```
where `xyz` is whatever service it previously said was already set (I think I had `lightdm` set previously or something like that).

Now you can reboot your Pi and see if it works
```
sudo shutdown -r now
```

fuzzievohs, 2020-02-12:
I've tried following the directions, but I can't get the octoscreen to run. Pi boots automatically, but to CLI, not octoscreen. I've checked and the octoscreen service is running. Any help would be greatly appreciate!

andreasnilsson, 2020-03-28:
> If you get here and are trying to make this work on a Raspberry Pi / Raspien ...
>
> I got mine to work by doing:
>
> ```
> sudo raspi-config
> ```
>
> go into "boot" options and set it to boot to a graphical [...]

You're the hero I needed! I had huge issues with getting my Raspberry pi 4B to display. But you're steps helped me almost get there.

I also needed to install `FBCP` once I did that it works perfect. Now I just need to update some configs since I have 320x480 screen.

roely15mol, 2020-10-03:
If i do sudo systemctl status octoscreen i get another error, any one that can help?

 octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-10-03 17:28:50 CEST; 17s ago
  Process: 778 ExecStartPost=/bin/bash /etc/octoscreen/disablescreenblank.sh 0 (code=exited, status=0/SUCCESS)
 Main PID: 757 (xinit)
    Tasks: 19 (limit: 2200)
   Memory: 23.2M
   CGroup: /system.slice/octoscreen.service
           ├─757 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor
           ├─758 /usr/lib/xorg/Xorg :0 -nolisten tcp -nocursor
           └─766 /usr/bin/OctoScreen

Oct 03 17:28:38 octopi xinit[757]:         (++) from command line, (!!) notice, (II) informational,
Oct 03 17:28:38 octopi xinit[757]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Oct 03 17:28:38 octopi xinit[757]: (==) Log file: "/var/log/Xorg.0.log", Time: Sat Oct  3 17:28:38 2020
Oct 03 17:28:38 octopi xinit[757]: (==) Using config directory: "/etc/X11/xorg.conf.d"
Oct 03 17:28:38 octopi xinit[757]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Oct 03 17:28:39 octopi xinit[757]: time="2020-10-03T17:28:39+02:00" level=info msg="OctoPrint's config file found: \"/home/pi/.octoprint/config.yaml\"" file=main.go func=main.readConfig lin
Oct 03 17:28:39 octopi xinit[757]: time="2020-10-03T17:28:39+02:00" level=info msg="Using \"http://localhost:5000\" as server address" file=main.go func=main.init.0 line=52
Oct 03 17:28:40 octopi xinit[757]: time="2020-10-03T17:28:40+02:00" level=debug msg="New background task started" file=common.go func="ui.(*BackgroundTask).Start" line=136
Oct 03 17:28:40 octopi xinit[757]: time="2020-10-03T17:28:40+02:00" level=error msg="unexpected status code: 404" file=ui.go func="ui.(*UI).loadSettings" line=194
Oct 03 17:28:50 octopi systemd[1]: Started OctoScreen.

JeffB42, 2021-01-06:
@ARCH-3D Is this still an issue?  Did you get your display working?

JeffB42, 2021-03-27:
Closed due to lack of response.

closed 2021-03-27 by JeffB42

Nbrooks639, 2022-04-19:
i still have not got my screen working per that last comment by another user.