Z-Bolt/OctoScreen #36: Missing 0.1mm Z movement

chilman408, 2019-10-15:
Hello,

I just started to use this setup.  It is very nice!  Everything works.

Just wanted to report that in the Movement Screen, Z-movement is missing 0.1mm steps.  (This stepping is available on Octoprint's webpage)

Also, in the Movement Screen, Z+ and Z- is reversed on the screen (Z+ should be on top).

Lastly, as a request, would it be possible to enable baby-stepping?  (less than 0.1mm movement?)  This will make this interface perfect for my use.

Thanks so much -

alexkv, 2019-10-15:
Try steps 0.1 mm in this version https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.3.5.
Position of Z+ and Z- is not a bug, its a feature :). It works for ultimaker-like printers, and as I understand you use prusa-like printer.
About baby-stepping, we will consider it in the future. And now you can try to configure it using OctoPrint custom commands. They display on the Control screen.

closed 2019-10-15 by alexkv

chilman408, 2019-10-16:
Just reporting back that the pre-release v2.3.5 works.  Although, cycling the moving stepping is this 1mm => 0.1mm => 5mm => 10mm.