Z-Bolt/OctoScreen #40: Why are z-axis control buttons reversed?

mdaneman, 2019-10-16:
For my printer on which the extruded moves along the z-axis, the z movement buttons are reversed. The down arrow moves it up and the up arrow moves it down. Confusing... and sometimes dangerous as you may accidentally ram the hot end into the build plate or workpiece. 
I can see how for a printer with a z-moving bed they would be correct. 
Can there be an configuration option to set the orientation of these buttons?

CorrieDivitt, 2019-11-07:
I'm using a Prusa i3 MK2/s and all axes are reversed. X, Y and Z.

marcusramberg, 2019-12-17:
It seems that the code is correct here, it's just the ui buttons for up/down z  that are switched around, so should be pretty easy to fix by renaming the images.

alexkv, 2019-12-25:
Try a new version https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.5.1

closed 2019-12-25 by alexkv