Z-Bolt/OctoScreen #42: Opinion of OctoScreen

ARCH-3D, 2019-10-18:
Hello alexkv,

I leave here my opinion about OctoScreen.
The design is simple and beautiful :) maybe a bit bigger the icons and the text and its perfect ;) 

Now some problems (with my specs, and other users in the same situation):
The Main Menu only fits correctly on my screen if i do a lot of extra steps: https://github.com/Z-Bolt/OctoScreen/issues/34#issuecomment-541835209 . But anyway....I see different resolutions when i change to print menu (when prints) and it doesnt fits correctly. Main Menu and Print Menu have different resolutions on my screen, cutting some parts on right side.
I also see that OctoScreen have problems with the connection on boot, and dont starts automatically, unless I do extra steps: https://github.com/Z-Bolt/OctoScreen/issues/34#issuecomment-541886187
Also bed and extruder temperature does not appear in main menu as in your post 
Sometimes it seems to block after a while with the printer off, maybe this situation: https://github.com/Z-Bolt/OctoScreen/issues/10
Some features are not necessary and are unusable buttons in my case (for example: move bed up and down). It would be nice if we could choose to eliminate this buttons
I see that the menu have space to four extruders, but i have the Prusa MMU2 and it have 5 filaments, it would be nice if could be 5 tools. 

I think that it will be a super fantastic app, but will be nice to solve this issues. Maybe very hard to solve :S i Dont Know but i send You my opinion. Thanks for the great Work that You are doing :)

My system:
-Raspberry Pi3b+
-Raspbian Octopi
-Velleman TFT 3.5" 320 x 480 - Resistive Touchscreen Display. GPIO

Cheers,

ljscz, 2019-11-28:
The problem with missing bed and extruders temperature on some screens is that the implementation works with max 4 extruders and the 5 "virtual" extruders of MMU2 defaults to "no info". 

I think that the best solution would be to show just one extruder based on the state of `profile.Extruder.SharedNozzle` which is `true` for MMU2. That solves the info screen problem but would also need to be used for screens where `/api/printer/tool` is used which returns temperatures for all virtual tools.

[patch.zip](https://github.com/Z-Bolt/OctoScreen/files/3903141/patch.zip)

ARCH-3D, 2019-11-29:
> The problem with missing bed and extruders temperature on some screens is that the implementation works with max 4 extruders and the 5 "virtual" extruders of MMU2 defaults to "no info".
>
> I think [...]

Thanks for your comment ljscz :) Yes i agree, the best solution would be to show just one extruder because are 5 shared tools by one nozzle. The attached file ("patch") that you send on your message is to solve this?

ljscz, 2019-11-29:
> Thanks for your comment ljscz :) Yes i agree, the best solution would be to show just one extruder because are 5 shared tools by one nozzle. The attached file ("patch") that you send on your message [...]

Yes the patch solves the problem. It is however changing not only the OctoScreen code but also the Octoprint go library so probably the real solution would have to be more complicated. This change at least works for me and makes the UI look good with MMU2.

ARCH-3D, 2019-11-29:
Ok ljscz Thanks, I'd like to try :) Can you give me some tips on how to use the patch? :)  thanks

ARCH-3D, 2019-11-29:
> Yes the patch solves the problem. It is however changing not only the OctoScreen code but also the Octoprint go library so probably the real solution would have to be more complicated. This change [...]

Ok ljscz Thanks, I'd like to try :) Can you give me some tips on how to use the patch? :) thanks

ljscz, 2019-11-29:
> Ok ljscz Thanks, I'd like to try :) Can you give me some tips on how to use the patch? :) thanks

You will have to build your own version of OctoScreen using the process described on the main page (very simple). Then just call `git apply patch.diff` before you start the `sudo make build`.

closed 2020-08-09 by JeffB42