Z-Bolt/OctoScreen #43: This Project is just straight up broken

FlyingT, 2019-10-19:


mvoss96, 2019-10-19:
Hi, i have it working on an pi 3b and 480x320 hdmi screen.
To scale to the screen i just had to set the correct resolution in raspbian

gdachs, 2019-10-20:
@FlyingT nonsense. This project works just fine for me. I use it on an Odroid C2 that is running Ubuntu. Attached is a 5" TFT via HDMI. So maybe your setup is broken and the fact that you tried to fix the problem by reinstalls shows only that you might be not experienced enough to even judge about the state of the project. On Linux there is no need to do any reinstalls to fix problems.

ursweiss, 2019-10-20:
Same problem with a 3.5" GPIO attached display, but spent much more time trying to get it to work. And I'm a Linux admin for 20 years. OK, luckily without having to spend much time with X11... 😁

I cannot get it to scale down in any way. Tried fb-turbo, fbcp, different OctoScreen and boot configs (framebuffer, hdmi_cvt options).

Finally tried [raspi2fb](https://github.com/AndrewFromMelbourne/raspi2fb). I could scale the console down, but as soon i start the OctoScreen service it switches back. After starting the raspi2fb service i do see that it scales down the screen from 800x600 to 480x320 in its service status.

As a test I installed OctoPrint-TFT now, which works with 480x320 displays out of the box and it works fine. But still would like to use OctoScreen.

x50arm, 2019-10-25:
I just picked up this touchscreen https://www.amazon.com/gp/product/B07N38B86S/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1  and after hours of reading pieces of information all over the internet I finally have it working.  This is my first experience with Raspberry Pi so this may not be the "right" way, but it worked for me.  I couldn't find all of this information in one place so I made a reference file.

Install Octopi
	Flash image to your storage media using "balenaEtcher".
	Using notepad++ edit the "octopi-wpa-supplicant" file for your wifi settings.

	
	Install storage media and boot up Raspberry Pi
	Get IP address of Raspberry Pi and connect to it using "Putty"
	

		
Install Octoscreen

	Dependencies install:
		
		Run: 
	
			sudo apt update
			
			sudo apt full-upgrade
		
			sudo apt-get install git build-essential xorg-dev xutils-dev x11proto-dri2-dev
			
			sudo apt-get install libltdl-dev libtool automake libdrm-dev
			
			sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils
						
			git clone https://github.com/ssvb/xf86-video-fbturbo.git
			
			cd xf86-video-fbturbo
			
			autoreconf -vi
			
			./configure --prefix=/usr
			
			make
			
			sudo make install
			
			sudo cp xorg.conf /etc/X11/xorg.conf
					

			
			
	Octoscreen install:
	
		Follow install instructions on Octoscreen Releases Page
		
		
	Edit Octoscreen Resolution
	
		Run:
		
			sudo nano /etc/octoscreen/config
			
			Edit OCTOSCREEN RESOLUTION as below:
				OCTOSCREEN_RESOLUTION=800x533
			
			ctrl-o
			
			enter
			
			ctrl-x
		
Install Touchscreen driver:

	Run:
		
		git clone https://github.com/Elecrow-keen/Elecrow-LCD35.git

		cd Elecrow-LCD35

		sudo ./Elecrow-LCD35

		sudo nano /etc/X11/xorg.conf.d/99-calibration.conf
		
		Go to the bottom of the text and hit "enter" to create a blank line.
		After that line type in the following lines:
			Section “Device” 
			  Identifier “touchscreen”
			  Driver “fbdev”
			  Option “fbdev” “/dev/fb0”
			EndSection
			
		ctrl-o
		
		enter
		
		ctrl-x
		
		
Install fbcp to "clone" the HDMI output:

	Run:	
		sudo apt-get -y install cmake
		
		git clone https://github.com/tasanakorn/rpi-fbcp
		
		cd rpi-fbcp
		
		mkdir build
		
		cd build
		
		cmake ..
		
		make
		
		sudo install fbcp /usr/local/bin/fbcp	
		
		
Make fbcp start on every boot:

	Run:	
		sudo nano /etc/rc.local

		-> Then, go at the end of file (but before the exit 0 ) and add a line where to find fbcp :

		/usr/local/bin/fbcp &	
		
		ctrl-o
		
		enter
		
		ctrl-x
		
		
Activate and setup HDMI:

	Run:	
		sudo nano /boot/config.txt

		Uncomment the following and change resolution
			framebuffer_width=800
			framebuffer_height=533

		add line
			hdmi_cvt 800 533 60 6 0 0 0		
		
		ctrl-o
		
		enter
		
		ctrl-x
		
		sudo reboot
		
		
		
Upon reboot touchscreen should be up and running.

CorrieDivitt, 2019-11-07:
> I just picked up this touchscreen https://www.amazon.com/gp/product/B07N38B86S/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1 and after hours of reading pieces of information all over the internet [...]

I just wanted to say x50arm thank you! I literally spent 10 hours trying different things to get the Octoprint-TFT plugin working and this one. Nothing worked, I tried 4 different sets of instructions and the best I could get was TouchUI working. Your guide actually got things working. I'm not quite sure which part of it was the saving grace but it actually worked. Thank you!!

x50arm, 2019-11-07:
> I just wanted to say @x50arm thank you! I literally spent 10 hours trying different things to get the Octoprint-TFT plugin working and this one. Nothing worked, I tried 4 different sets of instructions and the best I could get was TouchUI working. Your guide actually got things working. I'm not quite sure which part of it was the saving grace but it actually worked. Thank you!!
> 
> I was also wondering if there is ever any need to put the API key into the config file?

Your welcome, I spent many hours figuring this out.  Glad it helped you.  I did not put the api key in the config.  I did however go back and change my resolution to 800x600 to use all of the available area on my display.

MasterTim17, 2019-11-20:
hi @x50arm 
i followed all these helpful steps, but all i ended with is the following screen :(
[image]

x50arm, 2019-11-20:
> hi @x50arm
> i followed all these helpful steps, but all i ended with is the following screen :(

I’m new to the Raspberry Pi and have no clue what might be causing your problem.  Hopefully, someone will chime in that can help you troubleshoot.

mikekscholz, 2019-12-12:
@x50arm Having all of the information in one place is strictly forbidden.

jdeath, 2020-01-17:
@x50arm

Thanks! Your directions worked. I just had to retype the quotes in the /etc/X11/xorg.conf.d/99-calibration.conf file, because it did not like the forward/backword quotes I got from copy/paste your instructions .

clowrey, 2020-02-23:
> I just had to retype the quotes in the /etc/X11/xorg.conf.d/99-calibration.conf file, because it did not like the forward/backward quotes I got from copy/paste your instructions .

That was my problem!!! thanks for noticing the pasted quotes are of an incorrect type..

ssombra, 2020-03-06:
> I just picked up this touchscreen https://www.amazon.com/gp/product/B07N38B86S/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1 and after hours of reading pieces of information all over the internet [...]

> 
> Install Touchscreen driver:
> 
> ```
> Run:
> 	
> 	git clone https://github.com/Elecrow-keen/Elecrow-LCD35.git
> 	cd Elecrow-LCD35
> 	sudo ./Elecrow-LCD35
> 	sudo nano /etc/X11/xorg.conf.d/99-calibration.conf
> 

@x50arm  I have the same LCD as yours and having issues following the amazon instructions so I plan to try with your instructions later today. I see that you are using different drivers (_Elecrow-LCD35_ instead of _LCD-show git page_) and different LCD (_Elecrow-LCD35_ instead of _MHS35-show_) than the recommended in the amazon page, do you think that is the problem with the amazon instructions? or your fix has to do with the dependencies install?

x50arm, 2020-03-07:
@ssombra   I really couldn’t say.  Just posted what worked for me after many hours trying different things.

ssombra, 2020-03-07:
> @ssombra I really couldn’t say. Just posted what worked for me after many hours trying different things.

@x50arm Thank you very much! it worked! I spent several hours yesterday with no success! I just wish to understand why now works :) but I am a total ignorant on linux. Thanks again

Vahtera, 2020-03-27:
I have the same problem, but even @x50arm 's instructions aren't helping. If I leave the line "/usr/local/bin/fbcp &" out of /etc/rc.local, I can see Octoscreen UI, but it is the wrong resolution (ie, I can only see the top left of the UI).

When I add "/usr/local/bin/fbcp &" to the rc.local file (or start fbcp manually from the command line), screen goes to blank black and I can *barely* see a cursos in the top left corner. No UI, no terminal, nothing. 

What am I doing wrong?

jaburges, 2020-04-01:
Similar issue - no screen detected (which i'm reading on the screen connected to the pi)
how is the OS able to see the screen but octoscreen is not?

CorrieDivitt, 2020-04-01:
I had a working build and when I recently took an update for touch ui it broke and it can't detect the screen. Not sure if the newest version is incompatible

Oleniko1, 2020-04-23:
I had the same issue with the "No screen detected" error it is not a issue with octoscreen.
I found out that in my case it was in /etc/X11/xorg.conf.d/99-calibration.conf
the lines that u were supposed to copy to the file made it brake.
` Section “Device” 
      Identifier “touchscreen” 
      Driver “fbdev”
      Option “fbdev” “/dev/fb0”
EndSection`
After removing this from the config the "No screen detected" error is gone and octoprint is working just fine

Fretzy, 2020-04-27:
@Oleniko1 Thank you! I was having the same issue as @MasterTim17 and this fixed my problem immediately.

hrqiang, 2020-05-08:
@x50arm I follow you instruction. It displays perfectly. However, the touch screen doesn't work. 
I don't know this fbcp method how the touch actually works. But when I switch to without fbcp, I've no way to get the resolution to 320x480.

velocentric, 2020-05-21:
@x50arm thank you for compiling those instructions. It took me a few tries to get it working. I think the quotes copy function was the main issue. Unfortunately my Waveshare 3.5B V2 is displaying Octoscreen in the wrong colors, but after three days of just trying to see that damned octopus on the screen, This feels like progress.

abhi911kumar, 2020-09-09:
> I just picked up this touchscreen https://www.amazon.com/gp/product/B07N38B86S/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1 and after hours of reading pieces of information all over the internet [...]

Wonderful. Was struggling for the solution for so long and now your solution had helped. Thank you :1st_place_medal:

pjratl, 2020-10-13:
ok I am strugling here I followed @x50arm instructions on boot I see the logo screen flash but the octopus is off to the far right then the screen goes black with a small -- in the top left 

any ideas?

JeffB42, 2021-01-06:
@FlyingT Is this still an issue?  Did you get your display working?  Here are some set up guides:
https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display
and https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

FlyingT, 2021-01-06:
@JeffB42 No, never
Just switched to OctoDash, which scaled perfectly out-of-the-box

closed 2021-01-06 by FlyingT