Z-Bolt/OctoScreen #45: Question:. Does this work in Portrait mode?

chilman408, 2019-10-20:
If so, what is a quick way to make this happen?

alexkv, 2019-10-28:
The only way I see to make it happen is to develop a separate pack of screens for portrait mode. 
Unfortunately, it's no quick way.

AstroClimber, 2020-01-09:
Yes, but you will have to configure your touch screen in portrait mode, then modify the hdmi_cvt parameters, you can find some of the guide in Issue #72

JeffB42, 2021-03-27:
@chilman408 sorry, but it doesn't.  OctoScreen is set up into a 4 column x 3 row display layout.  It's written using GTK, which isn't a responsive design UI library (meaning, it doesn't automatically adjust its layout depending on the screen orientation).  This layout is pretty entranced in OctoScreen, and at this time I don't plan to add support for a portrait layout.

closed 2021-03-27 by JeffB42