Z-Bolt/OctoScreen #46: Bug in vesion 2.3.5

rferrari81, 2019-10-22:
hi, I installed version 2.3.5 but there is a bug.  if you press the menu move when the axes move, the app crashes and nothing moves

lele344, 2019-10-23:
ah ok, same here! ;(

AlmightyFireFly, 2019-10-28:
I have 5 printers with raspberry pi's with octoscreen installed.  All crash if i press z up in the move menu.  Right now i have rolled back to the previous version but did like the improvements in 2.3.5.  Is there a way to manually correct the issue?  Thank you in advance. 
Respectfully.

alexkv, 2019-10-28:
Hi there. 
I will upload the fix of this problem during the day.

AlmightyFireFly, 2019-10-28:
Thank you.  I really do like octoscreen and prefer it to the other interfaces I have tried.

rferrari81, 2019-10-28:
thanks, you should also post a very detailed guide on how to install it.  I managed to install it but I don't really know why I couldn't.

alexkv, 2019-10-29:
try this version: https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.3.6

closed 2019-10-29 by alexkv

rferrari81, 2019-10-29:
now it works, you should add time remaining to print and print from sd printer