Z-Bolt/OctoScreen #48: Problem with installation

corso1, 2019-10-24:
Im using Raspberry 3b+ and Raspberry 7" touchscreen display.
Try guide from github for .deb installation but without success.

When I get to this step: sudo dpkg -i octoscreen_2.3-5_armhf.deb
It respond with this: Failed to preset unit:
File /etc/systemd/system/display-manager.service already exists and is a symlink to /lib/systemd/system/lightdm.service.
/usr/bin/deb-systemd-helper:
error: systemctl preset failed on octoscreen.service: No such file or directory

And this is end of the way for me, please can someone help me with this instalation.
Thank you in advance.

gdachs, 2019-10-24:
It looks like you have already installed a complete desktop. Are you aware that this desktop would not appear anymore after you have installed OctoScreen?
OctoScreen tries to create a new symlink to its own service, but this is already there, created by your desktop. Just delete this symlink: `sudo rm /etc/systemd/system/display-manager.service` and try to install OctoScreen again.

corso1, 2019-10-24:
Now I got this respond...
Installing service /lib/systemd/system/lightdm.service
WARNING: /lib/systemd/system/lightdm.service is the selected default display manager but does not exist

gdachs, 2019-10-25:
New problem, new issue

corso1, 2019-10-25:
Problem solved thank you!

closed 2019-10-25 by corso1

florin03, 2020-03-08:
Hello.
I receive the same message after installing my lcd 3.5 tft driver 
https://www.aliexpress.com/item/32707058182.html?spm=a2g0s.9042311.0.0.27424c4dZvHESd

for the driver i use ;

wget http://www.waveshare.com/w/upload/0/00/LCD-show-170703.tar.gz
tar xvf LCD-show-*.tar.gz
cd LCD-show/
chmod +x LCD35-show
./LCD35-show

Thank you

icXuicXu, 2020-11-25:
Uninstall the desktop `sudo dpkg -r --force-depends lightdm` then try to install once more. Regards, 
icXu