Z-Bolt/OctoScreen #49: CR-10S Pro issues

corso1, 2019-10-25:
I have few issues when connect Creality CR-10S Pro with Octoscreen which running on Raspberry 3b+ with raspberry 7" touchscreen.

- First problem is extruding, when I try to extrude with "slow" command my extruder go crazy and start skipping. 
- Second is with move commands, when I try to move any axis I get this:
[image]
- Third issue, I dont see anything in print directory on octoscreen, sd card is pluged in printer.

corso1, 2019-10-25:
1. In retract/feed menu you have options slow/fast, Im using Marlin 2.0.
2. I have raspberry power suply, when I use home command it works.
3. Maybe its dumb question but how to upload gcode file if I use independently octoscreen on raspberry without octoprint on pc.
4. Server doesn’t respond, I dont know how to access octoprint.

EvilGremlin, 2019-10-25:
1. Clarified, looks like improper value then (or too fast for your stepper driver)
2. This shouldn't happen at all. If your board powered from PRi USB, remove that connection (remove fuse, cut wire/trace, whatever there is). Or maybe your PSU is not enough to run PRi with display.
3. Through webUI or from slicer.
4. Connect with SSH/telnet, then see what's wrong with webserver.
Anyway, at least describe your installation process (start from writing armbian image to SD). Else all you'll have is futile random guesses.

alexkv, 2019-10-28:
> Third issue, I dont see anything in print directory on octoscreen, sd card is pluged in printer.
OctoScreen doen't work with printer's SD cards.

JeffB42, 2021-03-27:
@corso1 I know this is an old bug.  If you are still using OctoScreen, is this an issue?  If it is, please update to 2.7.0, as it fixes a lot of bugs since 2.5.

closed 2021-04-17 by JeffB42