Z-Bolt/OctoScreen #51: Not sure how to configure screen

italocjs, 2019-10-31:
I managed to install the octoscreen after some issues (i had install a bunch of packages needed, but its fine now)  and the service is starting, but it closes on the error "no screen found".

I 'm using a screen model "waveshare 3.5a" which is 480x320px, the driver is installed and i can see the terminal in the screen (already changed the resolution to 800x533 as specified)

Where should i set the screen model, or something, so the octoscreen boots up?  i'm pretty lost at this point.

italocjs, 2019-10-31:
Also, i tried installing this thing ( https://github.com/ssvb/xf86-video-fbturbo/wiki/Installation ) but i got errors:

`pi@octopi:~/xf86-video-fbturbo $         autoreconf -vi
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal
configure.ac:38: error: must install xorg-macros 1.8 or later before running autoconf/autogen
configure.ac:38: the top level
autom4te: /usr/bin/m4 failed with exit status: 1
aclocal: error: echo failed with exit status: 1
autoreconf: aclocal failed with exit status: 1
pi@octopi:~/xf86-video-fbturbo $         ./configure --prefix=/usr
-bash: ./configure: No such file or directory
pi@octopi:~/xf86-video-fbturbo $         make
make: *** No targets specified and no makefile found.  Stop.
`

italocjs, 2019-10-31:
Managed to install, this xorg-macros someway didnt install, i executed sudo apt-get update --fix-missing and installed all again, now the octoscreen is botting up, but stuck at a boot loop

Pi boots, show terminal
octoscreen logo and wifi button appear for ~5
goes back to terminal
octoscreen logo and wifi button appear for ~5

all repeat,  printer is connected (i tought it could be it)

my terminal says:

```
Oct 31 14:22:23 octopi xinit[770]:         to make sure that you have the latest version.
Oct 31 14:22:23 octopi xinit[770]: Markers: (--) probed, (**) from config file, (==) default setting,
Oct 31 14:22:23 octopi xinit[770]:         (++) from command line, (!!) notice, (II) informational,
Oct 31 14:22:23 octopi xinit[770]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Oct 31 14:22:23 octopi xinit[770]: (==) Log file: "/var/log/Xorg.0.log", Time: Thu Oct 31 14:22:23 2019
Oct 31 14:22:23 octopi xinit[770]: (==) Using config file: "/etc/X11/xorg.conf"
Oct 31 14:22:23 octopi xinit[770]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Oct 31 14:22:23 octopi xinit[770]: modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/4.14.79-v7+
Oct 31 14:22:23 octopi xinit[770]: time="2019-10-31T14:22:23Z" level=info msg="OctoPrint's config file found: \"/home/pi/.octoprint/config.yaml\"" file=main.go func=mai
Oct 31 14:22:24 octopi xinit[770]: time="2019-10-31T14:22:24Z" level=debug msg="New background task started" file=common.go func="ui.(*BackgroundTask).Start" line=114
```

x50arm, 2019-10-31:
Try following what I posted here and see if it works: https://github.com/Z-Bolt/OctoScreen/issues/43

JeffB42, 2020-08-09:
@italocjs - I don't think OctoScreen will work with your screen.  It appears the absolute minimum resolution needed is 584x348.

closed 2020-08-09 by JeffB42

italocjs, 2020-08-09:
It works @JeffB42 , i followed #43.  sorry i forgot to update.

The reason i'm not using octodash anymore is because i cant print from USB/SD easily, when this feature is added i'll certaily test again.   Right now we are using some astrotouchs.