Z-Bolt/OctoScreen #52: [Request] pre compiled image

italocjs, 2019-10-31:
Hi, i would like to suggest the creation of pre compiled images, this would certainly help noobs like me, who never used linux and those stuff before and want to try this new screen software :D

alexkv, 2019-11-23:
Hi, it's a good idea. The main issue here is, that you need the different images for the different models of displays.

I have an image for the display HDMI LCD 4.3 (https://www.waveshare.com/product/modules/oleds-lcds/raspberry-pi-lcd/4.3inch-hdmi-lcd-b.htm): 
https://drive.google.com/open?id=1cCrXka1bQ-PvD1MBWoh4mqUMJE9ze7qS

Credentials: pi / R8uq=iJDx8fcqkYRvvM*

bvalentino1, 2020-01-23:
Great image! Thank you for that!
the only problem is that i dont have the password when i want to log in via SSH. So i cant install another screen. Could you make an image with the default raspberry password?

davidbrown-vsolvit, 2020-05-05:
I have the same waveshare 4.3 and can’t get octoscreen to work. The display keeps cycling between the Octoprint GUI and the octopi login prompt. I’ve been at this for two days and I’m stumped. 🤷🏼‍♂️

davidbrown-vsolvit, 2020-05-05:
> Hi, it's a good idea. The main issue here is, that you need different images for the different models of displays.
> 
> I have an image for the display HDMI LCD 4.3 (https://www.waveshare.com/product/modules/oleds-lcds/raspberry-pi-lcd/4.3inch-hdmi-lcd-b.htm):
> https://drive.google.com/open?id=1cCrXka1bQ-PvD1MBWoh4mqUMJE9ze7qS
> 
> Credentials: pi / R8uq=iJDx8fcqkYRvvM*

I’ll download your image tomorrow and give it a shot. 😎

Btw I do love octoscreen. It should be an octoprint supported plugin.

davidbrown-vsolvit, 2020-05-05:
> Hi, it's a good idea. The main issue here is, that you need different images for the different models of displays.
>
> I have an image for the display HDMI LCD 4.3 [...]

I downloaded your image and was wondering how I combine your image and an image for Octoprint, as your image doesn’t seem to contain Octoprint.