Z-Bolt/OctoScreen #53: Octoscreen loads then resets every 10 seconds

Hesit8or, 2019-11-01:
Having an Issue with Octoscreen restarting every 10 seconds, Any Ideas how to fix this ?

pi@octopi:~ $ systemctl status octoscreen.service
● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: activating (start-post) since Sat 2019-11-02 00:54:36 AEDT; 5s ago
 Main PID: 2088 (xinit); Control PID: 2109 (bash)
    Tasks: 20 (limit: 4915)
   CGroup: /system.slice/octoscreen.service
           ├─2088 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor
           ├─2089 /usr/lib/xorg/Xorg :0 -nolisten tcp -nocursor
           ├─2097 /usr/bin/OctoScreen
           └─control
             ├─2109 /bin/bash /etc/octoscreen/disablescreenblank.sh 0
             └─2110 sleep 10

Nov 02 00:54:36 octopi xinit[2088]:         (++) from command line, (!!) notice, (II) informational,
Nov 02 00:54:36 octopi xinit[2088]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Nov 02 00:54:36 octopi xinit[2088]: (==) Log file: "/var/log/Xorg.0.log", Time: Sat Nov  2 00:54:36 2019
Nov 02 00:54:36 octopi xinit[2088]: (==) Using config directory: "/etc/X11/xorg.conf.d"
Nov 02 00:54:36 octopi xinit[2088]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Nov 02 00:54:36 octopi xinit[2088]: time="2019-11-02T00:54:36+11:00" level=info msg="OctoPrint's config file found: \"/home/pi/.octoprint/config.yaml\"" file=main.go func=main.readConfig line=102
Nov 02 00:54:36 octopi xinit[2088]: time="2019-11-02T00:54:36+11:00" level=info msg="Using \"http://localhost:5000\" as server address" file=main.go func=main.init.0 line=52
Nov 02 00:54:37 octopi xinit[2088]: time="2019-11-02T00:54:37+11:00" level=debug msg="New background task started" file=common.go func="ui.(*BackgroundTask).Start" line=114
Nov 02 00:54:37 octopi xinit[2088]: time="2019-11-02T00:54:37+11:00" level=error msg="unexpected status code: 404" file=ui.go func="ui.(*UI).checkNotification" line=175
Nov 02 00:54:39 octopi xinit[2088]: time="2019-11-02T00:54:39+11:00" level=error msg="unexpected status code: 404" file=ui.go func="ui.(*UI).checkNotification" line=175

pi@octopi:~ $ journalctl -xe
Nov 02 00:50:54 octopi systemd[1]: Failed to start OctoScreen.
-- Subject: Unit octoscreen.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit octoscreen.service has failed.
-- 
-- The result is failed.
Nov 02 00:50:54 octopi systemd[1]: octoscreen.service: Unit entered failed state.
Nov 02 00:50:54 octopi systemd[1]: octoscreen.service: Failed with result 'exit-code'.
Nov 02 00:50:54 octopi systemd[1]: octoscreen.service: Service hold-off time over, scheduling restart.
Nov 02 00:50:54 octopi systemd[1]: Stopped OctoScreen.
-- Subject: Unit octoscreen.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit octoscreen.service has finished shutting down.
Nov 02 00:50:54 octopi systemd[1]: Starting OctoScreen...
-- Subject: Unit octoscreen.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit octoscreen.service has begun starting up.
Nov 02 00:50:54 octopi xinit[952]: X.Org X Server 1.19.2
Nov 02 00:50:54 octopi xinit[952]: Release Date: 2017-03-02
Nov 02 00:50:54 octopi xinit[952]: X Protocol Version 11, Revision 0
Nov 02 00:50:54 octopi xinit[952]: Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
Nov 02 00:50:54 octopi xinit[952]: Current Operating System: Linux octopi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l
Nov 02 00:50:54 octopi xinit[952]: Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x400
Nov 02 00:50:54 octopi xinit[952]: Build Date: 18 October 2017  04:55:30PM
Nov 02 00:50:54 octopi xinit[952]: xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support)
Nov 02 00:50:54 octopi xinit[952]: Current version of pixman: 0.34.0
Nov 02 00:50:54 octopi xinit[952]:         Before reporting problems, check http://wiki.x.org
Nov 02 00:50:54 octopi xinit[952]:         to make sure that you have the latest version.
Nov 02 00:50:54 octopi xinit[952]: Markers: (--) probed, (**) from config file, (==) default setting,
Nov 02 00:50:54 octopi xinit[952]:         (++) from command line, (!!) notice, (II) informational,
Nov 02 00:50:54 octopi xinit[952]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Nov 02 00:50:54 octopi xinit[952]: (==) Log file: "/var/log/Xorg.0.log", Time: Sat Nov  2 00:50:54 2019
Nov 02 00:50:54 octopi xinit[952]: (==) Using config directory: "/etc/X11/xorg.conf.d"
Nov 02 00:50:54 octopi xinit[952]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Nov 02 00:50:55 octopi xinit[952]: time="2019-11-02T00:50:55+11:00" level=info msg="OctoPrint's config file found: \"/home/pi/.octoprint/config.yaml\"" file=main.go func=main.readConfig line=102
Nov 02 00:50:55 octopi xinit[952]: time="2019-11-02T00:50:55+11:00" level=info msg="Using \"http://localhost:5000\" as server address" file=main.go func=main.init.0 line=52
Nov 02 00:50:55 octopi xinit[952]: time="2019-11-02T00:50:55+11:00" level=debug msg="New background task started" file=common.go func="ui.(*BackgroundTask).Start" line=114
Nov 02 00:50:56 octopi xinit[952]: time="2019-11-02T00:50:56+11:00" level=error msg="unexpected status code: 404" file=ui.go func="ui.(*UI).checkNotification" line=175
Nov 02 00:50:58 octopi xinit[952]: time="2019-11-02T00:50:58+11:00" level=error msg="unexpected status code: 404" file=ui.go func="ui.(*UI).checkNotification" line=175
Nov 02 00:51:00 octopi xinit[952]: time="2019-11-02T00:51:00+11:00" level=error msg="unexpected status code: 404" file=ui.go func="ui.(*UI).checkNotification" line=175
Nov 02 00:51:02 octopi xinit[952]: time="2019-11-02T00:51:02+11:00" level=error msg="unexpected status code: 404" file=ui.go func="ui.(*UI).checkNotification" line=175
Nov 02 00:51:04 octopi xinit[952]: time="2019-11-02T00:51:04+11:00" level=error msg="unexpected status code: 404" file=ui.go func="ui.(*UI).checkNotification" line=175
lines 1724-1771/1771 (END)

thealanshow, 2019-11-04:
I had this issue and noticed it does that because I wasn't logged in on the pi itself. I set the pi to autologin, then reassigned a startup script for the Octoscreen and poof that issue is gone!

bluecamel, 2020-01-10:
@thealanshow Would you mind elaborating on what you did?  I have this same issue.

My setup is a fresh octopi image.  I also ran the Adafruit TFT setup script and set hdmi_cvt per the documentation.  The pi boots up to a CLI login prompt.  Every few seconds, the screen flashes black when octoscreen restarts.

gazzer82, 2020-01-15:
Same problem here, TFT works, but I only ever get the login prompt and the screen flashing when the service restarts.

This is the status of the service:

`● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: failed (Result: protocol) since Wed 2020-01-15 15:45:20 GMT; 30s ago
  Process: 518 ExecStart=/usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor (code=exited, status=0/SUCCESS)
 Main PID: 518 (code=exited, status=0/SUCCESS)

Jan 15 15:45:20 raspberrypi systemd[1]: octoscreen.service: Service RestartSec=100ms expired, scheduling restart.
Jan 15 15:45:20 raspberrypi systemd[1]: octoscreen.service: Scheduled restart job, restart counter is at 9.
Jan 15 15:45:20 raspberrypi systemd[1]: Stopped OctoScreen.
Jan 15 15:45:20 raspberrypi systemd[1]: octoscreen.service: Start request repeated too quickly.
Jan 15 15:45:20 raspberrypi systemd[1]: octoscreen.service: Failed with result 'protocol'.
Jan 15 15:45:20 raspberrypi systemd[1]: Failed to start OctoScreen.
pi@raspberrypi:~ $ 
`

hunagyp, 2020-01-15:
Maybe you would get more information from octoscreen's log?
Did you switch it on?
/etc/octoscreen/config
OCTOSCREEN_LOG_FILE parameter

onyxsky02, 2020-03-06:
I am getting the same issue, this is what appears in the logs;

time="2020-03-06T15:10:43Z" level=info msg="OctoPrint's config file found: \"/home/pi/.octoprint/config.yaml\"" file=main.go func=main.readConfig line=102
time="2020-03-06T15:10:43Z" level=info msg="Using \"http://localhost:5000\" as server address" file=main.go func=main.init.0 line=52
time="2020-03-06T15:10:44Z" level=debug msg="New background task started" file=common.go func="ui.(*BackgroundTask).Start" line=136
time="2020-03-06T15:10:44Z" level=error msg="unexpected status code: 404" file=ui.go func="ui.(*UI).loadSettings" line=194

evilbobbins, 2020-03-08:
Same issue on a fresh install @thealanshow what were the steps you did to reassign the scrip to the pi user after the auto login change?

Heikoho, 2020-03-13:
Hi,
I have the same issue.
For 3 days I have been reinstalling my Octoprint with Octoscreen. Again and again mistakes.
Now I've finally completely reinstalled everything: First Octopi (version 16, with the current one somehow didn't work), then the display driver (5 "TFT), then Octoscreen according to the instructions in the GIT. And then the update to Version 17.
Now finally (almost) everything works. Octopi is running, the printer is recognized and I can control it and Octoscreen is running. At least every now and then.
But Octoscreen changes the picture almost exactly every 10-12 seconds. Looks like he's resetting. The octoscreen can be seen for a short time for almost 10 seconds and then it switches over again and very briefly shows the terminal lines ending with the Octopi login. Then he switches back to Octoscreen and it starts again.
I urgently ask for help, so unfortunately I can not use Octoscreen.
I hope I have expressed myself clearly to some extent. I have attached a screen image of the octoprint.log

Thank you

[image]

mythias41, 2020-05-08:
> Hi,
> I have the same issue.
> For 3 days I have been reinstalling my Octoprint with Octoscreen. Again and again mistakes.
> Now I've finally completely reinstalled everything: First Octopi (version 16, with the current one somehow didn't work), then the display driver (5 "TFT), then Octoscreen according to the instructions in the GIT. And then the update to Version 17.
> Now finally (almost) everything works. Octopi is running, the printer is recognized and I can control it and Octoscreen is running. At least every now and then.
> But Octoscreen changes the picture almost exactly every 10-12 seconds. Looks like he's resetting. The octoscreen can be seen for a short time for almost 10 seconds and then it switches over again and very briefly shows the terminal lines ending with the Octopi login. Then he switches back to Octoscreen and it starts again.
> I urgently ask for help, so unfortunately I can not use Octoscreen.
> I hope I have expressed myself clearly to some extent. I have attached a screen image of the octoprint.log
> 
> Thank you
> 
> P.S.:
> I'm just noticing that I don't have the File "plugin/zbolt_octoscreen" . Is that correct? Code 404 shows that a file is missing, doesn't it?
> 
> [image]

I'm currently having the same issue as you. Told the pi to boot to gui with auto login and still it switches between Octoscreen and the terminal. Were you able to fix this issue? Thanks in advance

k3nnethl, 2020-07-11:
> I am getting the same issue, this is what appears in the logs;
>
> time="2020-03-06T15:10:43Z" level=info msg="OctoPrint's config file found: "/home/pi/.octoprint/config.yaml"" file=main.go [...]

did you manage to fix it? i'm seeing the exact same error :c

michaelnoardo, 2020-09-05:
I managed to fix it reinstalling it, because i tried to install it before apt update and some dependencies were corrupted, that is why after 10 sec it was connection again, after just running the following commands it just fixed

sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils

Download the octoscreen plugin:
wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.5.1/octoscreen_2.5-1_armhf.deb
   
   Install octoscreen:
sudo dpkg -i octoscreen_2.5-1_armhf.deb

JeffB42, 2020-09-08:
Looks like this is fixed now/there's a know solutions?  I'm going to close this ticket.

closed 2020-09-08 by JeffB42