Z-Bolt/OctoScreen #56: OctoScreen crashing OctoPrint when printer not connected/powered on

rhukster, 2019-11-19:
I've been setting up my OctoPrint setup on a raspberry pi 3b and it looks to be working well.  However, when the printer is off and the pi + OctoScreen is on, I get these messages in the console continuously:

```
Changing monitoring state from "Offline" to "Detecting serial port"
Serial port list: []
Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
Failed to autodetect serial port, please set it manually.
...
```

After a few minutes, OctoPrint becomes unresponsive and requires a restart. When I stop the `octoscreen.service`, the messages stop and OctoPrint remains functional.

Is there any way to control the constant reconnecting attempts? or limit it somehow?  I don't want to have to boot or restart my pi every time I turn my printer on.

Cheers!

MatthewCroughan, 2019-11-21:
https://github.com/Z-Bolt/OctoScreen/issues/10
https://github.com/Z-Bolt/OctoScreen/issues/58
These issues are all related.

MatthewCroughan, 2019-11-21:
I've found this commit, so maybe we can modify the rate at least. https://github.com/Z-Bolt/OctoScreen/commit/44e0003133209c7aa0ccda9ac0ae62fd49d4baa4

MatthewCroughan, 2019-11-21:
https://github.com/Z-Bolt/OctoScreen/commit/282df5096517041d50b196202e35443a52691f40#diff-13f2cd524f6b9b072ac15e8394891e3d

This commit changes the aforementioned code to:
```go
ui.b = NewBackgroundTask(time.Second*2, ui.update)
```

I don't quite know what ui.update is, but I imagine doing it every 2 seconds is causing this memory leak. I'm going to higher the value and test it. @alexkv do you think you know what might be going on here now that I've found this?

rhukster, 2019-11-21:
I think adjusting the rate or having a config var for It would be useful. However, I think the real issue probably lies in that verifyConnection method. I’ve not looked but probably need to make sure it’s fully closing the connection.  It could also be on octoprint side, for example if the api method that is being called is not closing connections or has a memory leak.

MatthewCroughan, 2019-11-21:
If you see this log from #58, what you're saying would probably be the case.. I don't know any GO, so I'm taking this bit by bit. If Tornado is crashing then surely this is a problem with Octoprint, no? Either way, Octoprint should be robust enough to handle this. Maybe I'm being naive though.

```20.11.19 16:33:07 (+0000)  octoprint  2019-11-20 16:33:07,014 - tornado.general - ERROR - Uncaught exception
20.11.19 16:33:07 (+0000)  octoprint  Traceback (most recent call last):
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 775, in _server_request_loop
20.11.19 16:33:07 (+0000)  octoprint      ret = yield conn.read_response(request_delegate)
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/gen.py", line 1055, in run
20.11.19 16:33:07 (+0000)  octoprint      value = future.result()
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/concurrent.py", line 238, in result
20.11.19 16:33:07 (+0000)  octoprint      raise_exc_info(self._exc_info)
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/gen.py", line 307, in wrapper
20.11.19 16:33:07 (+0000)  octoprint      yielded = next(result)
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/http1connection.py", line 159, in _read_message
20.11.19 16:33:07 (+0000)  octoprint      max_bytes=self.params.max_header_size)
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 256, in read_until_regex
20.11.19 16:33:07 (+0000)  octoprint      self._try_inline_read()
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 711, in _try_inline_read
20.11.19 16:33:07 (+0000)  octoprint      pos = self._read_to_buffer_loop()
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 625, in _read_to_buffer_loop
20.11.19 16:33:07 (+0000)  octoprint      if self._read_to_buffer() == 0:
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 738, in _read_to_buffer
20.11.19 16:33:07 (+0000)  octoprint      chunk = self.read_from_fd()
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 1052, in read_from_fd
20.11.19 16:33:07 (+0000)  octoprint      chunk = self.socket.recv(self.read_chunk_size)
20.11.19 16:33:07 (+0000)  octoprint  MemoryError```

https://github.com/foosel/OctoPrint/issues/3343

MatthewCroughan, 2019-11-21:
We can do one of two things to verify this. We can use github.com/mcuadros/go-octoprint to see if we can replicate this issue with requests alone. Or we can use the older version of the code to see if it still happens.

MatthewCroughan, 2019-11-21:
Having a hard time reading GO, need to learn it to understand what's going on here. https://github.com/mcuadros/go-octoprint/blob/master/connection.go

I'm having a hard time understanding how OctoScreen tells octoprint.ConnectionRequest to do "autoconnect"

gdachs, 2019-11-21:
I have currently no possibility to check this issue myself, but if I would have I would check first whether the older background tasks gets terminated before the new one is started. You could just check with htop how much threads for OctoScreen are active when OctoPrint crashes. 
OctoScreen accesses OctoPrint via a REST API, this kind of API is stateless. So if OctoPrint crashes it is an OctoPrint problem.
It would not surprise me if all the old OctoScreen tasks are still active and OctoPrint needs some resources for every connection and doesn't work well with a too big amount of client connections.

MatthewCroughan, 2019-11-22:
@gdachs 

Indeed you're right, Octoscreen does accumulate threads.

[image]

MatthewCroughan, 2019-11-22:
The problem is that even if I kill these Octoscreen threads, or all of Octoscreen, this still leaves Octoprint completely broken and choking on MemoryError in Tornado until the end of time.

gdachs, 2019-11-22:
Obviously killing the threads doesn't release the resources used in OctoPrint. The right thing to do is to cleanly stop this threads in OctoScreen before a NewBackgroundTask is started.

MatthewCroughan, 2019-11-22:
@gdachs ui.b.Close() will do this after looking into b = BackgroundTask's methodson L39 of UI.go. I'm going to try and do something about it, testing now. But if you come up with a solution first let me know.

MatthewCroughan, 2019-11-22:
@gdachs someone in the #go-nuts freenode chat has pointed out https://github.com/Z-Bolt/OctoScreen/blob/845d348b565c55bb198e3bdf18c4956449690a54/ui/common.go#L148

```go
func (t *BackgroundTask) execute() {
	_, err := glib.IdleAdd(t.task)
	if err != nil {
		log.Fatal("IdleAdd() failed:", err)
	}
}
```

gdachs, 2019-11-22:
Sorry, I can't help you with this. I know many programming languages, but not this one.
But from my understanding that just outputs an error message and would not stop a thread.

MatthewCroughan, 2019-11-22:
Indeed, I also am struggling to understand GO. Trying my best ;)

gdachs, 2019-11-22:
If I would have the time, I would redo it in C++ ;)

gdachs, 2019-11-22:
I just looked into glib what IdleAdd is doing. I assume it is this function: [g_idle_add](https://developer.gnome.org/glib/stable/glib-The-Main-Event-Loop.html#g-idle-add).
So I would believe that it would be necessary to call the equivalent of this function beforehand: [g_idle_remove_by_data](https://developer.gnome.org/glib/stable/glib-The-Main-Event-Loop.html#g-idle-remove-by-data). What do you think?

MatthewCroughan, 2019-11-22:
@gdachs I can give that a go and compile it basically right now and test it, since I have the full setup all ready to go. I'll let you know.

MatthewCroughan, 2019-11-22:
Just to confirm. Since I really am not savvy enough yet to be operating in this arena, you think this:
```go
func (t *BackgroundTask) execute() {
	_, err := glib.IdleAdd(t.task)
	if err != nil {
		log.Fatal("IdleAdd() failed:", err)
	}
}
```

should become this:

```go
func (t *BackgroundTask) execute() {
        _, err := glib.IdleRemoveByData(t.task)
	_, err := glib.IdleAdd(t.task)
	if err != nil {
		log.Fatal("IdleAdd() failed:", err)
	}
}
```

gdachs, 2019-11-22:
Not sure that it is so easy. I would expect that t.task is the function, so you can't use that for IdleRemoveByData, as there is a data pointer expected. Don't know where the data comes from. Additionally the first call will always fail, not sure whether this is a problem. 
Reading the documentation again, it seems to be better if the t.task function would not hang, but instead exit with return code false, so that it gets automatically removed. No idea how to do this.

gdachs, 2019-11-22:
what about that:
`func (t *BackgroundTask) execute() {
       	if t.running {
		t.close()
	}
 	_, err := glib.IdleAdd(t.task)
	if err != nil {
		log.Fatal("IdleAdd() failed:", err)
	}
}`

gdachs, 2019-11-22:
shit, what has happened with the formating?

MatthewCroughan, 2019-11-22:
@gdachs you need to wrap it in three of these above and below ```

rhukster, 2019-11-22:
It would be great if @alexkv chimed in on this.. Clearly, it's a legitimate and significant issue, and he's the best person to tackle it.

MatthewCroughan, 2019-11-23:
@gdachs Tried compiling that. Let me know if you want SSH access to a machine where you can just modify ui.go and run `make build` to try things out.
"src/github.com/Z-Bolt/OctoScreen/ui/common.go:148:31: cannot call non-function t.close (type chan bool)"

MatthewCroughan, 2019-11-23:
Oh man.. It compiles if I change `t.close` to `t.Close` I wondered why the syntax highlighting was highlighting .close when not starting with an upper case.

MatthewCroughan, 2019-11-23:
It seems now to never attempt connecting on the serial. Weird.

gdachs, 2019-11-23:
I wouldn't like to be more involved in this issue currently. I even don't suffer from it. I use klipper firmware and my Odroid C2 and the controller of the printer are always powered on to the same time.

MatthewCroughan, 2019-11-23:
@gdachs On a sidenote, what's Klipper like? Anywhere I can add you to talk about that? :D

gdachs, 2019-11-23:
For Klipper it would be better to use the mailing list klipper@freelists.org, or #klipper on IRC

MatthewCroughan, 2019-11-23:
@alexkv Hooray! I'm so happy you've managed to find your way to this issue. Can you please tag it with any related commits by putting "#56" in the commit message, that way it'll show up in this issue. Thanks so much.

MatthewCroughan, 2019-11-23:
Additionally, I'm also available with a bunch of different pi's and a stable containerised octoprint + octoscreen setup that I've spent a lot of time on. So I can debug and build as much as possible.

MatthewCroughan, 2019-11-24:
Maybe @mcuadros could also comment, since he's the one who wrote the common.go threading logic.

MatthewCroughan, 2019-11-29:
@alexkv It's worth noting that whenever this issue gets fixed, I'm going to be making youtube videos about Octoscreen, as well as a few twitter posts. So I cannot wait.

MatthewCroughan, 2019-12-10:
@alexkv I've tested this in a wide range of cases now. It is not Octoscreen. This can be replicated with a simple polling of the Octoprint API, or by clicking the "Connect" button in a web interface. On the 3000th time a Pi 3 will map up to 2GB of virtual memory. On a 32 BIT linux OS this exceeds the boundary of what's possible to map, so the Tornado web framework that runs Octoprint promptly crashes.

Using a 64 bit OS on the pi alleviates this, but does not solve the root problem. Every time you click the "Connect" button, or Octosreen, or Curl poll the API, Octoprint will spawn a new thread that for some reason will never be closed. This means the garbage collector in python can never collect, leading to a memory leak.

https://github.com/foosel/OctoPrint/issues/3376

alexkv, 2019-12-25:
Try a new version https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.5.1

closed 2019-12-25 by alexkv

jchaltinner, 2020-02-06:
This is still happening to me. 
RPi 3b 1.2, Octopi 0.17.0, OctoPrint 1.3.12, OctoScreen 2.5.1 installed.

It's a constant scroll in the terminal window, and the "Unknown error" balloon popping up and going away over and over (the same message as OP @rhukster reported). It gets in the way of menu items (power, alerts, user settings), because it pops up on top of them even while using them. Eventually it gets to the point where I have to unplug the Pi. Trying to reset from the web interface, the OctoScreen interface, or even SSH no longer work.

When it gets the that point, I also get this on the OctoScreen every time it re-initializes in the loop: 

> Unexpected error: Get http://localhost:5000/api/connection: read tcp 127.0.0.1:58748->127.0.0.1:5000: read: connection reset by peer

It's also almost impossible to switch printer profiles even when I am connected to a printer, because almost immediately when I click disconnect, it tries to reconnect. I have to be really fast at changing the profile and hitting connect to the correct printer.

It goes away when I stop the OctoScreen service, and Octoprint works normally, so it's just when Octoscreen is running.

I thought my issue might be continuing after the 2.5.1 update because of the Malyan serial fix plugin, but I disabled that with no effect.

I get the easy solution is to turn on the printer... but quite often I want to manage files on the Pi, or work on the OctoScreen menu setup _without_ all the fans on my machines running. It's not a deal breaker, but wanted to make sure you know it's still happening even though you closed the issue.

nachosan100, 2020-02-25:
same here - just installed basically the same setup as @jchaltinner. AFter my printer turned off after finishing a print, i was bombarded by push messages (via nautilus and pushover plugin in octoprint) stating the printer to not be reachable...ill have to see if there is a fix -  if not, i have a problem and will have to dump it, as i will not keep printer unattended after it finishes. I hope someone has a fix for this headache.

advorsky73, 2020-03-16:
Same here.... Octoprint dies rather quick when the printer is not connected due to exzessive polling from this app.

justinmbrock, 2020-03-28:
@MatthewCroughan no, this is definitely an issue with OctoScreen. Why is it spamming OctoPrint to reconnect, even if you disconnect from the printer manually through the OctoPrint web interface?

BrewChef, 2020-04-03:
I have the same exact problem. Ocotoprint does not respond, when printer is "off" - when printer is on, Octoscreen reconnects and overrides Octoprint 100% of the time. It does however work, but it makes Octoprint more or less useless.

OevreFlataeker, 2020-09-02:
Is there still some activity on this bug? I came along the same way and am affected by it.

JeffB42, 2020-09-08:
@OevreFlataeker I think this bug might have been a different issue and was fixed (but you see similar results).  Could you open up a new bug and describe in detail what you are doing, and what you are seeing?  I'm not getting this issue, but I can look into after I release 2.6.

OevreFlataeker, 2020-09-08:
The problem is manifold as you have maybe heard already. The connection polls are creating more and more threads on the Octoprint side for reasons yet not fully understood. The first assumption from the Octoprint dev side was that the requests come in too fast so the tornado server blocks/hangs in those old requests. However it also doesn't seem that increasing the poll intervall changes this significantly. What I found was that the code for stopping the poll after 9 retries works as expected however there seems to linger one more bug. The systemd awaits a watchdog signal from OctoScreen which is only send during an actual connection poll so those are missing and lead to a restart of Octoprint which in turn restarts the connection polling etc. etc. I've fixed the bug and it works now like a charm. Let me know if you want a PR. It's just a one-liner in ui/ui.go:update() https://github.com/OevreFlataeker/OctoScreen

JeffB42, 2020-09-09:
@OevreFlataeker no need for a PR, I was working in that file and added that change.  BTW, I created a branching strategy doc in the wiki (https://github.com/Z-Bolt/OctoScreen/wiki/Branching-Strategy) and PRs should target the dev branch (not master).  Which btw, if you are interested in checking out, is https://github.com/Z-Bolt/OctoScreen/tree/2.6.0-dev.  It's a work in progress, and if you have the logging level set to info or warn, it outputs a lot to the console (and log file, if you have it configured).

OevreFlataeker, 2020-09-09:
Thanks for the update. Where can I read what exactly is being worked on in 2.6.0 (dev) please? (<- No frequent Github user here ;-) )