Z-Bolt/OctoScreen #58: Octoprint OUT OF MEMORY in Tornado due to excessive polling of serial port

MatthewCroughan, 2019-11-20:
After about an hour of being left on, Octoscreen causes the death of Octoprint as follows:
```
20.11.19 16:33:07 (+0000)  octoprint  2019-11-20 16:33:07,014 - tornado.general - ERROR - Uncaught exception
20.11.19 16:33:07 (+0000)  octoprint  Traceback (most recent call last):
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 775, in _server_request_loop
20.11.19 16:33:07 (+0000)  octoprint      ret = yield conn.read_response(request_delegate)
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/gen.py", line 1055, in run
20.11.19 16:33:07 (+0000)  octoprint      value = future.result()
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/concurrent.py", line 238, in result
20.11.19 16:33:07 (+0000)  octoprint      raise_exc_info(self._exc_info)
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/gen.py", line 307, in wrapper
20.11.19 16:33:07 (+0000)  octoprint      yielded = next(result)
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/http1connection.py", line 159, in _read_message
20.11.19 16:33:07 (+0000)  octoprint      max_bytes=self.params.max_header_size)
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 256, in read_until_regex
20.11.19 16:33:07 (+0000)  octoprint      self._try_inline_read()
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 711, in _try_inline_read
20.11.19 16:33:07 (+0000)  octoprint      pos = self._read_to_buffer_loop()
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 625, in _read_to_buffer_loop
20.11.19 16:33:07 (+0000)  octoprint      if self._read_to_buffer() == 0:
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 738, in _read_to_buffer
20.11.19 16:33:07 (+0000)  octoprint      chunk = self.read_from_fd()
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 1052, in read_from_fd
20.11.19 16:33:07 (+0000)  octoprint      chunk = self.socket.recv(self.read_chunk_size)
20.11.19 16:33:07 (+0000)  octoprint  MemoryError
```

MatthewCroughan, 2019-11-21:
I believe this must be related to this issue https://github.com/Z-Bolt/OctoScreen/issues/10

alexkv, 2019-12-25:
Try new version https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.5.1

closed 2019-12-25 by alexkv