Z-Bolt/OctoScreen #59: Z and Y axis Reversed

mdiharce, 2019-11-22:
Hello! First I want to thank you for this excellent project.
I am using a Tronxy X3 and in OctoPrint as its commands in OctoDroid (from the phone) the axis displacements work perfect, but when using the same commands in OctoScreen they are reversed, is there any tutorial to solve this?

Regards! And thank you very much in advance.

Daeraxa, 2019-11-22:
There is an issue open on the roadmap to fix issue #40. 
What version are you running? I know when I first installed it I had two reversed axes as I followed the readme guide which instructs to install version 2.2.1 rather than the latest 2.3.6. On update I believe only my X axis remains inverted.

mdiharce, 2019-11-23:
Good morning, the guide I used is https://github.com/Z-Bolt/OctoScreen (install it on November 20)

Daeraxa, 2019-11-23:
Well as of 8 hours ago a new version has been released (replacing the need for 2.3.6).
Instruction to update to the current version are in the release note:
https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.4.9

mdiharce, 2019-11-24:
Hello!  I updated to the new version and although it has some improvements,
the Home of the axes makes it perfect, but when moving the Z axis up or
down it is still inverted.

Daeraxa, 2019-11-25:
Unfortunately the only answer is to wait for the 2.5 release, issue #40 (of which this is a duplicate) is is on the roadmap - https://github.com/Z-Bolt/OctoScreen/projects/2

runkster428, 2021-01-26:
Install the Z-bolt Octoscreen plugin for Octoprint.  You will then have an option in Octoprint to reverse the z axis.  I had the same problem on my Ender 5 pro.
link - https://github.com/Z-Bolt/OctoPrint-Z-Bolt-OctoScreen

JeffB42, 2021-01-31:
@mdiharce , install the plug-in and you will have the option to reverse the z-direction.

closed 2021-01-31 by JeffB42