Z-Bolt/OctoScreen #61: No insight into SD-Card

Erik262, 2019-11-29:
Hi,
I realised that Octoscreen can't display files which are already on the SD-Card or has not been added through Octoprint.

gdachs, 2019-11-29:
Yes, this is true, but I don't think that this is an issue with OctoScreen. Just add the folder with the SD card to OctoPrint and OctoScreen will see it too. If you don't know how to do it, then look at this [discussion](https://community.octoprint.org/t/how-to-change-the-upload-folder-to-a-mounted-drive/6156).

mathieudupre, 2019-12-12:
Hi, almost same issue here that I didn't have with octoprint-tft
I just don't have any displayed file. A blank (dark) page.
I've checked the permissions, the location and the correct owner name (pi) of the uploads folder.
Any clue please?

Richardgriff, 2020-03-08:
I am also having the same issue. I have files on my SD card which can be seen in the octoprint webui however when i click the print button in Octoscreen no files are displayed.

dj-fiorex, 2020-10-22:
But how can i modify OctoScreen to show all SD files inside the printer?

JeffB42, 2021-01-05:
@Erik262 @Richardgriff @dj-fiorex this will be available in 2.7

JeffB42, 2021-03-27:
This has been implemented in 2.7.0

closed 2021-03-27 by JeffB42