Z-Bolt/OctoScreen #63: Installation problems

nettunoroberto, 2019-12-01:
hi friends, i have this problem during installation:

pi@octopi:~ $ sudo dpkg -i octoscreen_2.2-2_armhf.deb
Selecting previously unselected package octoscreen.
(Reading database ... 73940 files and directories currently installed.)
Preparing to unpack octoscreen_2.2-2_armhf.deb ...
Unpacking octoscreen (2.2-2) ...
dpkg: dependency problems prevent configuration of octoscreen:
 octoscreen depends on libgtk-3-0 (>= 3.21.5); however:
  Version of libgtk-3-0:armhf on system is 3.14.5-1+deb8u1rpi1rpi1g.

dpkg: error processing package octoscreen (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 octoscreen

gdachs, 2019-12-01:
Is there a special reason why you install a so old version of OctoScreen? The current version is 2.4.6.
Did you do sudo apt update, sudo apt dist-upgrade beforehand?

nettunoroberto, 2019-12-01:
sudo dpkg -i octoscreen_2.4-9_armhf.deb
Selecting previously unselected package octoscreen.
(Reading database ... 73888 files and directories currently installed.)
Preparing to unpack octoscreen_2.4-9_armhf.deb ...
Unpacking octoscreen (2.4-9) ...
dpkg: dependency problems prevent configuration of octoscreen:
 octoscreen depends on libgtk-3-0 (>= 3.21.5); however:
  Version of libgtk-3-0:armhf on system is 3.14.5-1+deb8u1rpi1rpi1g.

dpkg: error processing package octoscreen (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 octoscreen

gdachs, 2019-12-01:
You don't mention which operating system you are using. But anyway, you got the reason why the install fail. OctoScreen depends on  libgtk-3 in a version greater than your current installed version.
First way to solve this problem is to install the required libgtk-3 version. Ask the operating system vendor how to do this.
Second way is to build the OctoScreen package on your system yourself. This is described on OctoScreens first github page.
Generally this is **NOT** an OctoScreen issue. It is just impossible to provide OctoScreen packages for all operating systems that are able to run Octoscreen. 
As an example, I use OctoScreen on an Odroid C2 with Ubuntu. For that I have always to build the package myself. It works just fine.

nettunoroberto, 2019-12-01:
i have : 
OctoPrint version : 1.3.9
OctoPi version    : 0.14.0

Daeraxa, 2019-12-01:
Follow this from start to finish, it is an edit of complied guide written by somebody else for installing OctoScreen on an OctoPi with a generic 3.5" TFT touch screen - https://pastebin.com/bJQmkcTt

@gdachs - They are installing the old version (2.2) because that is the instruction on the README page and most people will just be going through it verbatim, it needs to be updated to 2.4.9

gdachs, 2019-12-01:
@nettunoroberto you still didn't answer my first question, whether you updated your OS with apt, but this not really important anymore, as your OctoPI installation is way too old. You claim that you are using 0.14.0, where 0.17.0 is the current version. This explains the too old libgtk-3 package.
@Daeraxa OctoScreen is a one man show, not a company with a support department and other staff. 
For me I am much more thankful that @alexkv is using his spare time to fix the bugs, instead of keeping the install instructions up to date. 
For me it is just common sense to take installation instructions with a grain of salt and automatically look up the current versions of anything.

Daeraxa, 2019-12-01:
@gdachs Hey, I'm not complaining that it isn't up to date, I fully understand the issues. I'm just explaining that is why the old version keeps being installed by people. For many people into 3d printing, this foray into the world of Linux and open source is entirely new to them and it is quite a daunting new world. What some people think is obvious is absolutely not for others.

nettunoroberto, 2019-12-01:
I make the sd with the latest octoprint images and I know how to reinstall all the packages. thanks for the information.

nettunoroberto, 2019-12-01:
I managed to install octoscreen but I would like to change the resolution because I have a 7 inch display

Daeraxa, 2019-12-01:
Check the readme section for configuration. You can set resolution via OCTOPRINT_RESOLUTION in the config file in /etc/octoscreen/config

nettunoroberto, 2019-12-01:
ok, tnx i have solve

gdachs, 2019-12-02:
@nettunoroberto then you could close this issue.

nettunoroberto, 2019-12-02:
Tnx for all

closed 2019-12-02 by nettunoroberto

tiagomaricate, 2020-08-15:
> hi friends, i have this problem during installation:
>
> pi@octopi:~ $ sudo dpkg -i octoscreen_2.2-2_armhf.deb
> Selecting previously unselected package octoscreen.
> (Reading database ... 73940 files and [...]

So no fix yet???

gdachs, 2020-08-16:
@tiagomaricate there is nothing broken. So no fix is required.

tiagomaricate, 2020-08-16:
> @tiagomaricate there is nothing broken. So no fix is required.

How can I force to install last version of gtk?
I tried update and the last version it install is always the 3.14.

gdachs, 2020-08-16:
@tiagomaricate that is just an issue with your operating system. This is the wrong place to discuss this. I suppose you use OctoPi so check whether you have installed the newest version and ask them for support.

JeffB42, 2020-08-18:
@tiagomaricate also, a few tips:
1) I noticed that you are installing octoscreen_2.2-2.  The current release is 2.5.1 and I suggest installing that instead.
2) I also suggest following Chris's video - https://www.youtube.com/watch?v=OJ59hXSyBoI&t=76s.  I followed along and didn't run into any issues.