Z-Bolt/OctoScreen #64: Debian Buster Install from source

ShohninDmitriy, 2019-12-02:
pi@rpi3router:~ $ git clone https://github.com/Z-Bolt/OctoScreen.git
Cloning into 'OctoScreen'...
remote: Enumerating objects: 69, done.
remote: Counting objects: 100% (69/69), done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 3267 (delta 28), reused 49 (delta 23), pack-reused 3198
Receiving objects: 100% (3267/3267), 3.46 MiB | 251.00 KiB/s, done.
Resolving deltas: 100% (1461/1461), done.
pi@rpi3router:~ $ cd OctoScreen
pi@rpi3router:~/OctoScreen $ make build
mkdir -p /home/pi/OctoScreen/build
docker build \
        --build-arg IMAGE=golang:1.9-stretch \
        --build-arg GO_TAGS=gtk_3_22 \
        -t mcuadros/octoprint-tft-build:stretch . \
        && \
docker run -it --rm \
        -v /home/pi/OctoScreen/build/stretch:/build \
        mcuadros/octoprint-tft-build:stretch \
        make build-internal
ERRO[0000] failed to dial gRPC: cannot connect to the Docker daemon. Is 'docker daemon' running on this host?: dial unix /var/run/docker.sock: connect: permission denied
ERRO[0000] Can't add file /home/pi/OctoScreen/.git/index to tar: io: read/write on closed pipe
ERRO[0000] Can't close tar writer: io: read/write on closed pipe
error during connect: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.40/build?buildargs=%7B%22GO_TAGS%22%3A%22gtk_3_22%22%2C%22IMAGE%22%3A%22golang%3A1.9-stretch%22%7D&cachefrom=%5B%5D&cgroupparent=&cpuperiod=0&cpuquota=0&cpusetcpus=&cpusetmems=&cpushares=0&dockerfile=Dockerfile&labels=%7B%7D&memory=0&memswap=0&networkmode=default&rm=1&session=osyijix1vtpfudcoi5x6qkj07&shmsize=0&t=mcuadros%2Foctoprint-tft-build%3Astretch&target=&ulimits=null&version=1: context canceled
make: *** [Makefile:52: STRETCH] Error 1
pi@rpi3router:~/OctoScreen $

gdachs, 2019-12-03:
did you not notice that you post is nearly unreadable? Anway, you didn't follow the instructions. You didn't use the commands that install docker.

ShohninDmitriy, 2019-12-04:
> 
> 
> did you not notice that your post is nearly unreadable? Anway, you didn't follow the instructions. You didn't use the commands that install docker.

you are wrong, everything is done according to the instructions and it does not work

Linux rpi3router 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Nov 30 17:53:27 2019 from 192.168.0.2
pi@rpi3router:~ $ curl -fsSL get.docker.com -o get-docker.sh
pi@rpi3router:~ $ sh get-docker.sh
# Executing docker install script, commit: f45d7c11389849ff46a6b4d94e0dd1ffebca32c1
Warning: the "docker" command appears to already exist on this system.

If you already have Docker installed, this script can cause trouble, which is
why we're displaying this warning and provide the opportunity to cancel the
installation.

If you installed the current Docker package using this script and are using it
again to update Docker, you can safely ignore this message.

You may press Ctrl+C now to abort this script.
+ sleep 20
+ sudo -E sh -c apt-get update -qq >/dev/null
+ sudo -E sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
+ sudo -E sh -c curl -fsSL "https://download.docker.com/linux/raspbian/gpg" | apt-key add -qq - >/dev/null
Warning: apt-key output should not be parsed (stdout is not a terminal)
+ sudo -E sh -c echo "deb [arch=armhf] https://download.docker.com/linux/raspbian buster stable" > /etc/apt/sources.list.d/docker.list
+ sudo -E sh -c apt-get update -qq >/dev/null
+ [ -n  ]
+ sudo -E sh -c apt-get install -y -qq --no-install-recommends docker-ce >/dev/null
+ sudo -E sh -c docker version
Client: Docker Engine - Community
 Version:           19.03.5
 API version:       1.40
 Go version:        go1.12.12
 Git commit:        633a0ea
 Built:             Wed Nov 13 07:37:22 2019
 OS/Arch:           linux/arm
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          19.03.5
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.12.12
  Git commit:       633a0ea
  Built:            Wed Nov 13 07:31:17 2019
  OS/Arch:          linux/arm
  Experimental:     false
 containerd:
  Version:          1.2.10
  GitCommit:        b34a5c8af56e510852c35414db4c1f4fa6172339
 runc:
  Version:          1.0.0-rc8+dev
  GitCommit:        3e425f80a8c931f88e6d94a8c831b9d5aa481657
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683
If you would like to use Docker as a non-root user, you should now consider
adding your user to the "docker" group with something like:

  sudo usermod -aG docker pi

Remember that you will have to log out and back in for this to take effect!

WARNING: Adding a user to the "docker" group will grant the ability to run
         containers which can be used to obtain root privileges on the
         docker host.
         Refer to https://docs.docker.com/engine/security/security/#docker-daemon-attack-surface
         for more information.
pi@rpi3router:~ $ sudo mc

pi@rpi3router:~ $ git pull https://github.com/Z-Bolt/OctoScreen.git
fatal: not a git repository (or any of the parent directories): .git
pi@rpi3router:~ $ git clone https://github.com/Z-Bolt/OctoScreen.git
Cloning into 'OctoScreen'...
remote: Enumerating objects: 69, done.
remote: Counting objects: 100% (69/69), done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 3267 (delta 28), reused 49 (delta 23), pack-reused 3198
Receiving objects: 100% (3267/3267), 3.46 MiB | 251.00 KiB/s, done.
Resolving deltas: 100% (1461/1461), done.
pi@rpi3router:~ $ cd OctoScreen
pi@rpi3router:~/OctoScreen $ make build
mkdir -p /home/pi/OctoScreen/build
docker build \
        --build-arg IMAGE=golang:1.9-stretch \
        --build-arg GO_TAGS=gtk_3_22 \
        -t mcuadros/octoprint-tft-build:stretch . \
        && \
docker run -it --rm \
        -v /home/pi/OctoScreen/build/stretch:/build \
        mcuadros/octoprint-tft-build:stretch \
        make build-internal
ERRO[0000] failed to dial gRPC: cannot connect to the Docker daemon. Is 'docker daemon' running on this host?: dial unix /var/run/docker.sock: connect: permission denied
ERRO[0000] Can't add file /home/pi/OctoScreen/.git/index to tar: io: read/write on closed pipe
ERRO[0000] Can't close tar writer: io: read/write on closed pipe
error during connect: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.40/build?buildargs=%7B%22GO_TAGS%22%3A%22gtk_3_22%22%2C%22IMAGE%22%3A%22golang%3A1.9-stretch%22%7D&cachefrom=%5B%5D&cgroupparent=&cpuperiod=0&cpuquota=0&cpusetcpus=&cpusetmems=&cpushares=0&dockerfile=Dockerfile&labels=%7B%7D&memory=0&memswap=0&networkmode=default&rm=1&session=osyijix1vtpfudcoi5x6qkj07&shmsize=0&t=mcuadros%2Foctoprint-tft-build%3Astretch&target=&ulimits=null&version=1: context canceled
make: *** [Makefile:52: STRETCH] Error 1
pi@rpi3router:~/OctoScreen $

gdachs, 2019-12-04:
Anyway, your post is still nearly unreadable, why don't you fix this? And why don't you react on the error messages you got and fix it? "dial unix /var/run/docker.sock: connect: permission denied"

This is not an OctoScreen issue. If you really have docker installed, then the user that you use for installing OctoScreen has no access to the docker socket.

ShohninDmitriy, 2019-12-04:
tried even on a clean image to install Docker and still does not work

closed 2019-12-06 by ShohninDmitriy