Z-Bolt/OctoScreen #66: Touch Screen Calibration

tjsc5f, 2019-12-08:
I have installed the adafruit 3.5" TFT and have changed the resolution to 800x533 in both the boot/config.txt file and the octoscreen/config file per the install instructions.  The scaling on the screen appears to be correct.  I ran the adafruit touch screen calibration tool, and everything seems to be working properly there, however when I try to touch buttons on the octoscreen interface, the presses are way off the target button.  I have to touch the top right corner to select the bottom right Network button on the home screen.  From the system info screen, pressing on the back button in the lower right corner actually activates the shutdown button in the lower left corner of the screen.  Any idea what I should do to fix this?

Daeraxa, 2019-12-08:
I had to edit /boot/config.txt and change the dtoverlay line to include rotation. dtoverlay=elecrow35a,**rotate=270**,speed=22000000,fps=60

Rook0316, 2019-12-08:
I spent a few days on this. I have the 2441 screen from Adafruit. If you rotated it 90 degrees you need to add - Option "TransformationMatrix" "0 1 0 -1 0 1 0 0 1" to your 40-libinput.conf file in /usr/share/X11/xorg.conf directory. You may need to use a different transformation matrix depending on your screen rotation.

This is the tutorial that really helped me. https://www.instructables.com/id/Rotate-Raspberry-Pi-Display-and-Touchscreen/

tjsc5f, 2019-12-08:
I rotated 270 degrees and had to use 0 -1 1 1 0 0 0 0 1 for the matrix, but that got it working correctly.  Thank you so much.

closed 2019-12-08 by tjsc5f