Z-Bolt/OctoScreen #68: Custom actions not show on Actions Page

kizza42, 2019-12-15:
I use the [Custom Control editor](https://plugins.octoprint.org/plugins/customControl/) plugin to inject custom Gcode commands in to Octoprint. They show nicely in the web interface but not all are displayed on OctoScreen, Would love to be able to trigger the extra commands from there.

[image]
[image]
[image]

mkutscher, 2020-01-15:
There seems to be a check which only allows single line command commands (or scripts):
https://github.com/Z-Bolt/OctoScreen/blob/master/ui/control.go#L70

BlueHoundInc, 2020-01-23:
The commands also won't show in the menu if there is only one command with blank lines after it.

rikvandersteen, 2021-01-31:
Is it already possible to add custom gcode   action buttons on the touch screen from octoscreen

JeffB42, 2021-02-28:
The issue is that OctoScreen has limited screen real estate.  When you give a command a long name, OctoScreen displays the name, there isn't enough room, and then the screen is expanded.  See all of the issues people having (in general) on 3.5" 640x480 displays, and their screen expanding and shifting on them.  One thing I could do is truncate the string to a max number of characters.

JeffB42, 2021-03-27:
> The commands also won't show in the menu if there is only one command with blank lines after it.

this was fixed by @thebeline and was fixed in the 2.7.0 release.

closed 2021-03-27 by JeffB42

thebeline, 2021-03-27:
#love