Z-Bolt/OctoScreen #72: FEATURE REQUEST: Support for 2.8" PiTFT

kieppie, 2019-12-27:
I'm making use of an old but quite common "XC9022" - aka. [Adafruit PiTFT - 320x240 2.8" TFT + Resistive Touchscreen](https://www.adafruit.com/product/1601) with the 4 GPIO buttons (#'s 23, 22, 27 & 18 - tested OK) populated.

Managed to get the screen firing up well & touchscreen calibrated using the Adafruit scripts/guides:

-  https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi?view=all
  -  https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/adafruit-pitft.sh

Unfortunately OctoScreen does not play/scale nice, and I've tried resorting to the `fbcp` "hack", but not getting great results:

-  https://learn.adafruit.com/running-opengl-based-games-and-emulators-on-adafruit-pitft-displays?view=all
  -  https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/pitft-fbcp.sh

from what I can tell, fbcp also adversely affects the use of the touchscreen interface, so that's a no-go

Is there a possible (simple) solution to this, and/or would it be possible to support this common piece of hardware in a future build?

AstroClimber, 2020-01-01:
Not sure if this will help to resolve your issues, I'm using the [PiTFT 2.8" 2423 Capacitive Screen](https://www.adafruit.com/product/2423). 

1. I have successfully configure my screen on RPi 3B with [Adafruit Easy Installation guide](https://learn.adafruit.com/adafruit-2-8-pitft-capacitive-touch/easy-install-2). I used the HDMI mirror option and set the screen to be landscape (option 1 or 3).

2. I following the instruction on https://github.com/Z-Bolt/OctoScreen:
`wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.5.1/octoscreen_2.5-1_armhf.deb
sudo dpkg -i octoscreen_2.5-1_armhf.deb`

3. Go in `/etc/octoscreen/config` and add 320x240 to the resolution configuration parameter as,
`OCTOSCREEN_RESOLUTION=320x240`

4.Go into `/boot/config.txt`, and edit the `hdml_cvt` line to,
`hdmi_cvt=660 360 60 1 0 0 `

5.`\sudo reboot`

6. Optional: If octoscreen does not start automatically, add `systemctl start octoscreen &` to the file `/etc/rc.local` right before the line `exit 0`. Note the `&` seems to be crucial to make it work, I tried without it and it failed to start OctoScreen. See Issue #34  for more infor on this.

**Remaining issue:** After rotating the PiTFT 90 or 270 degree to be landscape, the OctoScreen show that all the features are displayed properly, but when I touch them, the buttons seem to still think the screen is in the portrait 0degree orientation. I don't know if this is the PiTFT driver issue or a configuration in OctoScreen. Any insight or able to duplicate my problem will be helpful. Thanks.

AstroClimber, 2020-01-01:
> **Remaining issue:** After rotating the PiTFT 90 or 270 degree to be landscape, the OctoScreen show that all the features are displayed properly, but when I touch them, the buttons seem to still think the screen is in the portrait 0 degree orientation. So far, this [article](https://learn.adafruit.com/adafruit-2-8-pitft-capacitive-touch/easy-install-2) is what I found regarding to how HDMI and the touch screen rotations are two different things. I don't know if this is the PiTFT driver issue or a configuration in OctoScreen. Any insight or able to duplicate my problem will be helpful. Thanks.

OK, from the link posted in the quotation here, I found the parameters  that help to flip the PiTFT around. For the 2.8" Capacitive, they are
`Name:   pitft28-capacitive
Info:   Adafruit PiTFT 2.8" capacitive touch screen
Load:   dtoverlay=pitft28-capacitive,=
Params: speed                    Display SPI bus speed
        rotate                   Display rotation {0,90,180,270}
        fps                      Delay between frame updates
        debug                    Debug output level {0-7}
        touch-sizex              Touchscreen size x (default 240)
        touch-sizey              Touchscreen size y (default 320)
        touch-invx               Touchscreen inverted x axis
        touch-invy               Touchscreen inverted y axis
        touch-swapxy             Touchscreen swapped x y axis`

To make the update, add these two lines in `/boot/config.txt` and reboot. You may need to play around to get the X/Y flipping correct to the screen orientation you set.
`dtoverlay=pitft28-capacitive,rotate=90,speed=32000000,fps=20
dtoverlay=pitft28-capacitive,touch-swapxy=true,touch-invx=true`

Could be similar to the resistive one.

AstroClimber, 2020-01-02:
> To make the update, add these two lines in `/boot/config.txt` and reboot. You may need to play around to get the X/Y flipping correct to the screen orientation you set.
> `dtoverlay=pitft28-capacitive,rotate=90,speed=32000000,fps=20`
> `dtoverlay=pitft28-capacitive,touch-swapxy=true,touch-invx=true`

Apparently the instruction in that link is not accurate. I have to rotate 270 degree and by separating the `dtoverlay` into two separate line actually mess things up. I have to comment out the default `dtoverlay` lines and add the following (and ignore the speed and fps setting since there is 79 character max for the `dtoverlay` line apparently) and Voila, it works!
`#dtoverlay=pitft28-capacitive,rotate=270,speed=64000000,fps=30
  dtoverlay=pitft28-capacitive,rotate=270,touch-swapxy=true,touch-inx=true`

AstroClimber, 2020-01-02:
In case you want to control the backlight of the PiTFT after shutting down (it stays on after RPi is shutdown), check this [document](https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/backlight-control) out from Adafruit