Z-Bolt/OctoScreen #73: FEATURE REQUEST: Load/Unload filament length

illmatic-gr, 2019-12-30:
Hi,
the "laod filament" command calls G0 E600.0 F5000 and G0 E150.0 F500.
My Marlin default "EXTRUDE_MAXLENGTH " is 200 so I get the error too long extrusion prevented.
The same with unloading command. I think it is configured for bowden?
Is there a way to configure the length and speed for loading/unloading with octoscreen?

Btw, great work!
The interface is rebooting nearly every minute or two. Is there a logfile which could help you for debugging? Pi3 & Pi7"screen

kvandt, 2019-12-31:
I have the same rebooting issue. RPi3+ & 3.5" GPIO display. Printing itself is stable and continues without issue during UI restarts.

hamster65, 2020-04-13:
Where can I change the length for filament loading?
Same problem with rebooting for me.

rglory, 2020-04-25:
Looking into code values for load/unload are hardcoded, looks like at the moment you can only fix this by configuring your own buttons

nicedevil007, 2020-05-07:
Can anyone help me out with this?
I was able to load filament but not to unload it? It just won't do anything :/

JeffB42, 2021-03-27:
This was fixed in 2.7.0 which was just released.

closed 2021-03-27 by JeffB42