Z-Bolt/OctoScreen #79: Screen resolution or Scaling changes in the print menu.

AlmightyFireFly, 2020-01-16:
Everything was working very well.
Then I upgraded from version 2.3 to 2.5.
The install seems normal until I go to print, Then the buttons no longer fit on the screen.
I have a 800x480 5" touch screen.
I am on the latest octopi with all updates installed via ssh.
Is there something that would cause the resolution to change when accessing the print menu?
Is there something I can configure?

AlmightyFireFly, 2020-01-18:
[image]
[image]

chilman408, 2020-01-26:
I have the exact same issue here but can't find a solution...

chilman408, 2020-01-30:
Just for people waiting for a fix on this.  This fork addressed the problem.

https://github.com/mkutscher/OctoScreen/releases

Protoncek, 2020-02-01:
I wrote about this problem some time ago in issue #50... the problem is too long filename.  It's a bug, to be corrected in next version.

chilman408, 2020-02-03:
> I wrote about this problem some time ago in issue #50... the problem is too long filename. It's a bug, to be corrected in next version.

Thanks for the update...  Didn't know about this limitation.

rwaspe, 2020-05-24:
I posted a fix here: https://github.com/Z-Bolt/OctoScreen/issues/126

JeffB42, 2021-03-27:
@AlmightyFireFly is this still an issue? Please look over https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen. Also, v2.7.0 was just released. I suggest installing the latest as it fixes a lot of bugs.

closed 2021-04-17 by JeffB42