Z-Bolt/OctoScreen #80: May I try and replicate OctoScreen's functionality in Python?

MatthewCroughan, 2020-01-16:
Hey @alexkv I'm just asking for permission to take the entirety of OctoScreen and its concepts and try and replicate that in Python so that it's easier for people to contribute to! Hopefully this also means there are less issues as I personally will understand it a bit better and be able to fix some of the issues more rapidly. All credit will be given in the form of "Inspired by Z-Bolt's fork of OctoPrint-TFT https://github.com/Z-Bolt/OctoScreen"

The reason I am asking for permission despite the GPLV3 license is that I wish to use all of the SVG assets you've created for OctoScreen and feel it would be rude not to ask! Thanks for the work done.

hunagyp, 2020-01-17:
Very good idea, I hope you get the "greenlight" to do this! :)

alexkv, 2020-01-17:
Hi @MatthewCroughan, it's an awesome idea, feel free to use OctoPrint codes as you see fit. 

Actually I've already thought about it, as I'm also more familiar with Python and JS than with golang. While researching this question, I concluded that it, even more, promising to use Electrone/JS stack, as GTK has a lot of restrictions in UX/UI. Moreover, I already started the development of the new version on Electron/ReactJS but postponed it due to lack of time.

So if you are going to migrate OctoScreen to another stack, I can upload what I've already done in a separate project. I believe together we can achieve much better results.

rapzak, 2020-01-18:
+1 for python :) - maybe kivy?

steviehs, 2020-01-18:
Maybe you should join forces with the [Octodash Project](https://github.com/UnchartedBull/OctoDash) so there will be finally one working touch ui? As far as I understand, it uses electron/Angular...

JeffB42, 2021-03-27:
@MatthewCroughan Why reinvent the wheel?  I think your time would be better spent working on this (existing) project rather than starting a new project which emulates it.  It you do decide to create a new project, please don't call it OctoScreen, as that would cause confusion.

closed 2021-03-27 by JeffB42

MatthewCroughan, 2021-03-27:
@JeffB42 I ended up not doing anything. Though OctoDash seems to be a great project. The developer on that cannot manage their dependencies well and it has runtime update functionality which makes it hard to package/distribute, since it can break itself at runtime.