Z-Bolt/OctoScreen #82: OctoScreen resets every few minutes (octoscreen.service: Watchdog timeout)

morgendagen, 2020-01-20:
OctoScreen restarts every few minutes. Below is a combined log of /var/log/syslog and octoprint.log.

```
==> /var/log/syslog <==
Jan 20 22:59:43 octopi xinit[1240]: time="2020-01-20T22:59:43+01:00" level=info msg=4477 file=print_status.go func="ui.(*printStatusPanel).updateJob" line=277
Jan 20 22:59:45 octopi xinit[1240]: time="2020-01-20T22:59:45+01:00" level=info msg=4479 file=print_status.go func="ui.(*printStatusPanel).updateJob" line=277
Jan 20 22:59:55 octopi systemd[1]: octoscreen.service: Watchdog timeout (limit 10s)!
Jan 20 22:59:55 octopi systemd[1]: octoscreen.service: Killing process 1240 (xinit) with signal SIGABRT.
Jan 20 22:59:55 octopi xinit[1240]: (EE)
Jan 20 22:59:55 octopi xinit[1240]: (EE) Backtrace:
Jan 20 22:59:55 octopi xinit[1240]: (EE)
Jan 20 22:59:55 octopi xinit[1240]: (EE) Received signal 6 sent by process 1, uid 0
Jan 20 22:59:55 octopi xinit[1240]: (EE)
Jan 20 22:59:55 octopi xinit[1240]: Fatal server error:
Jan 20 22:59:55 octopi xinit[1240]: (EE) Caught signal 6 (Aborted). Server aborting
Jan 20 22:59:55 octopi xinit[1240]: (EE)
Jan 20 22:59:55 octopi xinit[1240]: (EE)
Jan 20 22:59:55 octopi xinit[1240]: Please consult the The X.Org Foundation support
Jan 20 22:59:55 octopi xinit[1240]: #011 at http://wiki.x.org
Jan 20 22:59:55 octopi xinit[1240]:  for help.
Jan 20 22:59:55 octopi xinit[1240]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
Jan 20 22:59:55 octopi xinit[1240]: (EE)
Jan 20 22:59:55 octopi systemd[1]: octoscreen.service: Killing process 1241 (Xorg) with signal SIGABRT.
Jan 20 22:59:55 octopi systemd[1]: octoscreen.service: Killing process 1252 (OctoScreen) with signal SIGABRT.
Jan 20 22:59:55 octopi systemd[1]: octoscreen.service: Main process exited, code=killed, status=6/ABRT
Jan 20 22:59:55 octopi xinit[1240]: (EE) Server terminated with error (1). Closing log file.
Jan 20 22:59:55 octopi systemd[1]: octoscreen.service: Failed with result 'watchdog'.
Jan 20 22:59:55 octopi systemd[1]: octoscreen.service: Service RestartSec=100ms expired, scheduling restart.
Jan 20 22:59:55 octopi systemd[1]: octoscreen.service: Scheduled restart job, restart counter is at 4.
Jan 20 22:59:55 octopi systemd[1]: Stopped OctoScreen.
Jan 20 22:59:55 octopi systemd[1]: Starting OctoScreen...
Jan 20 22:59:55 octopi xinit[1529]: X.Org X Server 1.20.4
Jan 20 22:59:55 octopi xinit[1529]: X Protocol Version 11, Revision 0
Jan 20 22:59:55 octopi xinit[1529]: Build Operating System: Linux 4.15.0-48-generic armv8l Raspbian
Jan 20 22:59:55 octopi xinit[1529]: Current Operating System: Linux octopi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l
Jan 20 22:59:55 octopi xinit[1529]: Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=600 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
Jan 20 22:59:55 octopi xinit[1529]: Build Date: 05 June 2019  12:49:54PM
Jan 20 22:59:55 octopi xinit[1529]: xorg-server 2:1.20.4-1+rpt1 (https://www.debian.org/support)
Jan 20 22:59:55 octopi xinit[1529]: Current version of pixman: 0.36.0
Jan 20 22:59:55 octopi xinit[1529]: #011Before reporting problems, check http://wiki.x.org
Jan 20 22:59:55 octopi xinit[1529]: #011to make sure that you have the latest version.
Jan 20 22:59:55 octopi xinit[1529]: Markers: (--) probed, (**) from config file, (==) default setting,
Jan 20 22:59:55 octopi xinit[1529]: #011(++) from command line, (!!) notice, (II) informational,
Jan 20 22:59:55 octopi xinit[1529]: #011(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Jan 20 22:59:55 octopi xinit[1529]: (==) Log file: "/var/log/Xorg.0.log", Time: Mon Jan 20 22:59:55 2020
Jan 20 22:59:55 octopi xinit[1529]: (==) Using config directory: "/etc/X11/xorg.conf.d"
Jan 20 22:59:55 octopi xinit[1529]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Jan 20 22:59:55 octopi xinit[1529]: modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/4.19.75-v7+
Jan 20 22:59:56 octopi xinit[1529]: time="2020-01-20T22:59:56+01:00" level=info msg="OctoPrint's config file found: \"/home/pi/.octoprint/config.yaml\"" file=main.go func=main.readConfig line=102
Jan 20 22:59:56 octopi xinit[1529]: time="2020-01-20T22:59:56+01:00" level=info msg="Using \"http://localhost:5000\" as server address" file=main.go func=main.init.0 line=52
Jan 20 22:59:56 octopi xinit[1529]: time="2020-01-20T22:59:56+01:00" level=info msg="Found API key at \"/home/pi/.octoprint/config.yaml\" file" file=main.go func=main.init.0 line=58
Jan 20 22:59:57 octopi xinit[1529]: time="2020-01-20T22:59:57+01:00" level=debug msg="New background task started" file=common.go func="ui.(*BackgroundTask).Start" line=136

==> octoprint.log <==
2020-01-20 22:59:57,333 - tornado.access - WARNING - 404 POST /api/plugin/zbolt_octoscreen (127.0.0.1) 14.02ms

==> /var/log/syslog <==
Jan 20 22:59:57 octopi xinit[1529]: time="2020-01-20T22:59:57+01:00" level=error msg="unexpected status code: 404" file=ui.go func="ui.(*UI).loadSettings" line=194

==> octoprint.log <==
2020-01-20 22:59:57,351 - tornado.access - WARNING - 404 POST /api/plugin/zbolt_octoscreen (127.0.0.1) 12.57ms

==> /var/log/syslog <==
Jan 20 22:59:57 octopi xinit[1529]: time="2020-01-20T22:59:57+01:00" level=info msg="Printing a job" file=ui.go func="ui.(*UI).verifyConnection" line=169
Jan 20 22:59:57 octopi xinit[1529]: time="2020-01-20T22:59:57+01:00" level=debug msg="New background task started" file=common.go func="ui.(*BackgroundTask).Start" line=136
Jan 20 22:59:57 octopi xinit[1529]: time="2020-01-20T22:59:57+01:00" level=info msg=4490 file=print_status.go func="ui.(*printStatusPanel).updateJob" line=277
Jan 20 22:59:59 octopi xinit[1529]: time="2020-01-20T22:59:59+01:00" level=info msg=4493 file=print_status.go func="ui.(*printStatusPanel).updateJob" line=277
Jan 20 23:00:01 octopi xinit[1529]: time="2020-01-20T23:00:01+01:00" level=info msg=4495 file=print_status.go func="ui.(*printStatusPanel).updateJob" line=277
Jan 20 23:00:03 octopi xinit[1529]: time="2020-01-20T23:00:03+01:00" level=info msg=4497 file=print_status.go func="ui.(*printStatusPanel).updateJob" line=277
Jan 20 23:00:05 octopi xinit[1529]: time="2020-01-20T23:00:05+01:00" level=info msg=4498 file=print_status.go func="ui.(*printStatusPanel).updateJob" line=277
Jan 20 23:00:07 octopi systemd[1]: Started OctoScreen.
Jan 20 23:00:07 octopi xinit[1529]: time="2020-01-20T23:00:07+01:00" level=info msg=4501 file=print_status.go func="ui.(*printStatusPanel).updateJob" line=277
Jan 20 23:00:09 octopi xinit[1529]: time="2020-01-20T23:00:09+01:00" level=info msg=4503 file=print_status.go func="ui.(*printStatusPanel).updateJob" line=277
```

rhukster, 2020-01-23:
I get this also. Running latest.

synman, 2020-01-26:
does this only happen when your printer is not connect?  I'm noticing some oddities with OctoScreen when my printer is powered down.

It appears to trigger OctoPrint's auto detection / connection logic in 2 second intervals and periodically shuts itself (OctoScreen) down and restarts.   

I'm guessing its trying to do auto-connection logic of some kind.  I haven't looked at the code yet, but guess I'll have to.

I discovered this while researching an OOM issue with OctoPrint.   I haven't confirmed it yet, but this polling may be exhausting OctoPrint's native heap (burning thru threads).

synman, 2020-01-26:
as suspected, in ui.go

```
func (m *UI) update() {
	if m.connectionAttempts > 8 {
		m.s.putOnHold()
		return
	} else if m.UIState == "splash" {
		m.connectionAttempts += 1
	} else {
		m.connectionAttempts = 0
	}

	if m.OctoPrintPlugin {
		m.checkNotification()
		m.loadSettings()
	}

	m.verifyConnection()
```

and in verifyConnection:

```
var errMercyPeriod = time.Second * 10

func (ui *UI) verifyConnection() {

	ui.sdNotify("WATCHDOG=1")

	newUiState := "splash"
	splashMessage := "Initializing..."

	s, err := (&octoprint.ConnectionRequest{}).Do(ui.Printer)
	if err == nil {
		ui.State = s.Current.State
		switch {
		case s.Current.State.IsOperational():
			newUiState = "idle"
		case s.Current.State.IsPrinting():
			newUiState = "printing"
		case s.Current.State.IsError():
			fallthrough
		case s.Current.State.IsOffline():
			if err := (&octoprint.ConnectRequest{}).Do(ui.Printer); err != nil {
				newUiState = "splash"
				splashMessage = "Loading..."
			}
		case s.Current.State.IsConnecting():
			splashMessage = string(s.Current.State)
		}
	} else {
		if time.Since(ui.t) > errMercyPeriod {
			splashMessage = ui.errToUser(err)
		}

		newUiState = "splash"
		Logger.Debugf("Unexpected error: %s", err)
	}

	defer func() { ui.UIState = newUiState }()

	ui.s.Label.SetText(splashMessage)

	if newUiState == ui.UIState {
		return
	}

	switch newUiState {
	case "idle":
		Logger.Info("Printer is ready")
		ui.Add(IdleStatusPanel(ui))
	case "printing":
		Logger.Info("Printing a job")
		ui.Add(PrintStatusPanel(ui))
	case "splash":
		ui.Add(ui.s)
	}
}
```

Ultimately leads to:
```
2020-01-26 14:13:36,249 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-01-26 14:13:36,263 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-01-26 14:13:38,274 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-01-26 14:13:38,288 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-01-26 14:13:40,247 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-01-26 14:13:40,265 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-01-26 14:13:42,233 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-01-26 14:13:42,234 - octoprint - ERROR - Exception on /api/connection [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1276, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1295, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/connection.py", line 67, in connectionCommand
    printer.connect(port=port, baudrate=baudrate, profile=printerProfile)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/printer/standard.py", line 264, in connect
    self._comm = comm.MachineCom(port, baudrate, callbackObject=self, printerProfileManager=self._printerProfileManager)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 605, in __init__
    self.sending_thread.start()
  File "/usr/lib/python2.7/threading.py", line 736, in start
    _start_new_thread(self.__bootstrap, ())
error: can't start new thread
2020-01-26 14:13:42,241 - tornado.access - ERROR - 500 POST /api/connection (127.0.0.1) 41.66ms
```

boom

synman, 2020-01-26:
let me know if this needs to be tracked as a separate issue if the OP is experiencing this with a "connected" printer.

Only way to recover is to restart octoprint.  it's a pretty nasty bug IMHO.   OctoScreen should leave connection logic up to OctoPrint (and its plugins) IMHO.

mdaneman, 2020-01-27:
I get something like this too with connected printer.  I think if Octoscreen doesn't hear back from Octoprint (or maybe from the printer) within a certain timeout, it restarts or goes into some funny mode temporarily.  I notice this if I'm uploading a file while a print it going on or in other cases where the printer (or maybe octoprint) is doing a long command.

synman, 2020-01-27:
The main UI thread has a 2 second refresh interval.  When fired, it checks for connectivity to 1.) octoprint itself and 2.) if octoprint is connected to a printer.  In the case where 1 is true but 2 is false, it instructs OctoPrint to connect to a printer.

After about 15 minutes of this (thrash) OctoPrint exhausts its native heap and can no longer create ANY new threads.  The octoprint instance is unusable at this point and the only way to recover is to restart it.

I believe it is at this point that OctoScreen starts receiving 500s for various API calls from the Octoprint instance (you can see one of them in my logs above).   

There are two additional timers at play here too.  The first is the "mercy" counter.  After 8 (failed) attempts, this triggers the retry button to show up on the OctoScreen UI.   The second is watchdog, and I believe (have not yet confirmed yet) he is a basic heartbeat thread triggered every two seconds via the main UI update thread (within verifyConnection).  

I have worked around this scenario with some very basic code commenting as I see no point in delegating printer connection responsibilities to OctoScreen itself.   Octoprint already has (extensive) built-in connection logic and with help of a plugin such as PortLister it can easily enumerate and connect to a new printer via a filesystem watcher (looks for new USB port attachments at a system level).

So, work around this, download the source and make to "hacks" to ui/ui.go:

first in` verifyConnection()`
```
...	
	case s.Current.State.IsOffline():
//			if err := (&octoprint.ConnectRequest{}).Do(ui.Printer); err != nil {
			newUiState = "splash"
			splashMessage = "Loading..."
//			}
...
```

and second in update()
```
...
func (m *UI) update() {
//	if m.connectionAttempts > 8 {
//		m.s.putOnHold()
//		return
//	} else if m.UIState == "splash" {
//		m.connectionAttempts += 1
//	} else {
//		m.connectionAttempts = 0
//	}

	if m.OctoPrintPlugin {
		m.checkNotification()
		m.loadSettings()
	}
...
```

problem sovled :)

synman, 2020-01-27:
regarding long running commands / file uploads / etc ... this too is a very valid callout and plausible condition.

There are certain activities that OctoPrint must perform on its main (looper) thread.  It will not respond to API, or other types of requests, during these operations.

My hunch is this is what is causing watchdog to kick in.   The connection status request towards the top of verifyConnection() is likely blocked waiting for a response.  Since the timer for update() is likely inactive (need to verify but I'm 99% it is) watchdog is not getting an WATCHDOG=1  refresh.

Within watchdog it likely looks for WATCHDOG=1 and resets it to something else when present.  The problem likely lies in the situation where WATCHDOG=1 is NOT present and watchdog does something silly like kill xinit because it thinks OctoScreen / X Windows is hung.

synman, 2020-01-27:
here we go:  

https://www.freedesktop.org/software/systemd/man/systemd.service.html#WatchdogSec=

Looks like we can adjust this to some MUCH higher value or disable it entirely by modifying OctoScreen's systemctl settings.

morgendagen, 2020-01-27:
> does this only happen when your printer is not connected? I'm noticing some oddities with OctoScreen when my printer is powered down.

No, the logs were captured while printing.

synman, 2020-01-27:
your issue is definitely watchdog... it's right there in your logs:
```
Jan 20 22:59:55 octopi systemd[1]: octoscreen.service: Watchdog timeout (limit 10s)!
Jan 20 22:59:55 octopi systemd[1]: octoscreen.service: Killing process 1240 (xinit) with signal SIGABRT.
```

IMHO, 10 seconds is a bit aggressive.  I bumped my watchdog timeout up to 300 seconds.   May I recommend something no less than 120?

**pi@:** `sudo nano /lib/systemd/system/octoscreen.service`

change:

`WatchdogSec=10s`

to:

`WatchdogSec=120s`

CTRL-X , Y

**pi@:** `sudo systemctl daemon-reload`

synman, 2020-01-27:
I went ahead and took a look at print_status.go, specifcally it's `updateJob()` method:
```
func (m *printStatusPanel) updateJob() {

	s, err := (&octoprint.JobRequest{}).Do(m.UI.Printer)
	if err != nil {
		Logger.Error(err)
		return
	}

	file := "<i>not-set</i>"
	if s.Job.File.Name != "" {
		file = s.Job.File.Name
		file = strings.Replace(file, ".gcode", "", -1)
		file = strEllipsisLen(file, 35)
	}

	m.file.Label.SetLabel(file)
	m.pb.SetFraction(s.Progress.Completion / 100)

	var timeSpent, timeLeft string
	switch s.Progress.Completion {
	case 100:
		timeSpent = fmt.Sprintf("Completed in %s", time.Duration(int64(s.Job.LastPrintTime)*1e9))
		timeLeft = ""
	case 0:
		timeSpent = "Warming up ..."
		timeLeft = ""
	default:
		Logger.Info(s.Progress.PrintTime)
		e := time.Duration(int64(s.Progress.PrintTime) * 1e9)
		r := time.Duration(int64(s.Progress.PrintTimeLeft) * 1e9)
		timeSpent = fmt.Sprintf("Time: %s", e)
		timeLeft = fmt.Sprintf("Left: %s", r)
	}

	m.time.Label.SetLabel(timeSpent)
	m.timeLeft.Label.SetLabel(timeLeft)
}
```

Line 277 is `Logger.Info(s.Progress.PrintTime)` ... a simple timestamp is logged as we see in your log file.

The meaty piece of this method is actually up at the very top:
```
s, err := (&octoprint.JobRequest{}).Do(m.UI.Printer)
```
This is an Octoprint API call, specifically /api/job (http://docs.octoprint.org/en/master/api/job.html#retrieve-information-about-the-current-job)

Something is causing a delay in response to this command from time to time.   Go with my prior recommendation of bumping up WatchdogSec to 2 minutes and I'm going to guess your problem goes away.

If it doesn't then there is something more pervasive happening here, likely beyond the control of OctoScreen.

boardthatpowder, 2020-01-31:
I'm experiencing the same issue.  I updated the WatchdogSec as suggested to 120s, but still experiencing the same issue.

boardthatpowder, 2020-01-31:
As suggested in ticket https://github.com/Z-Bolt/OctoScreen/issues/75, installing the https://github.com/Z-Bolt/OctoPrint-Z-Bolt-OctoScreen OctoPrint plugin fixed it for me.

synman, 2020-02-01:
that 404 is non-issue.   If you read the code you'll see it merely logs the condition and moves on.

If you look within the plug-in it references it exposes two basic settings (that are anemic) and a notifications interface that is 100% passive.

adrianmihalko, 2020-03-04:
When printer is not connected, instead of restarting app every x seconds and showing Initalizing... screen...it would be better to show offline screen and watch USB connection events with udev?

spawnrider, 2020-05-11:
+1 
Same issue for me...

JeffB42, 2021-03-27:
@morgendagen 2.7.0 was just release. Please update to the latest version, as some issues pertaining to watchdog were fixed.

closed 2021-03-27 by JeffB42