Z-Bolt/OctoScreen #84: Screen offset problem

epetit91, 2020-01-26:
Hello,
I encounter a small problem when I launch a print from Octoscreen.
When I press the Print icon, the screen shifts to the right (see image 1).
When the printing is finished, the screen remains offset and I no longer have access to certain commands (see image 2).
If I start printing from the Octoprint web interface, everything works perfectly and I do not observe any lag on the Octoscreen screen.
My server is a Raspberry PI 3B + with a 3.5 "PiTFT screen.
Octoscreen is in version 2.5 and Octoprint in version 1.3.12.
Has anyone ever observed this problem?
Thank you
[image]
[image]

chilman408, 2020-01-27:
I have the exact same issue.

This issue is same as issue #79

daimon82, 2020-03-06:
I have the same problem, at first its ok, but every time I stop a print from octoscreen this happens.

skibum-za, 2020-05-24:
Same issue for me. Using the official Raspberry Pi Touchscreen. Initially works correctly, then shifts things over. Makes it impossibe to use as you then can't access the options you need. So you can press the 'Print' button, but are then unable to actually select a print.

rwaspe, 2020-05-24:
I believe it is due to long file names as described in #126.  I posted a fix in that thread.  See if it helps.
https://github.com/Z-Bolt/OctoScreen/issues/126

skibum-za, 2020-05-27:
Thanks that fixed it for me. Clearly found the wrong thread mentioning the issue ;)

JeffB42, 2020-12-04:
@epetit91 Could you follow the steps listed in the wiki (located at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) and then post a reply to this issue whether the steps listed work and the issue is resolved?  I'd like to start closing out all the display-related issues.

If it doesn't work,
1) make sure you use a Raspberry Pi 3B or Raspberry Pi 4
2) start with a fresh install of OctoPi
3) please take a picture of the screen

Thanks, 

Jeff

JeffB42, 2021-01-06:
@epetit91 Is this still an issue?  Did you get your display working?

epetit91, 2021-01-06:
Hello,
Sorry for my late response.
I did not make the modification. Now I'm choosing shorter filenames to make sure the display doesn't shift.

Thank you

Eric

closed 2021-02-01 by JeffB42