Z-Bolt/OctoScreen #88: No screens found

McPrince96, 2020-02-09:
Hi,

Octoscreen doesn't want to start and the X.org X server's log report says this: 

`[  1026.249]
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[  1026.249] Build Operating System: Linux 4.15.0-48-generic armv8l Raspbian
[  1026.250] Current Operating System: Linux octopi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l
[  1026.250] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=533 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27$
[  1026.250] Build Date: 05 June 2019  12:49:54PM
[  1026.250] xorg-server 2:1.20.4-1+rpt1 (https://www.debian.org/support)
[  1026.250] Current version of pixman: 0.36.0
[  1026.250]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[  1026.250] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[  1026.251] (==) Log file: "/var/log/Xorg.0.log", Time: Sun Feb  9 15:12:24 2020
[  1026.252] (==) Using config file: "/etc/X11/xorg.conf"
[  1026.252] (==) Using config directory: "/etc/X11/xorg.conf.d"
[  1026.252] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[  1026.254] Parse error on line 8 of section InputClass in file /etc/X11/xorg.conf.d/99-calibration.conf
        The Section keyword requires a quoted string to follow it.
[  1026.254] (EE) Problem parsing the config file
[  1026.254] (EE) Error parsing the config file
[  1026.254] (EE)
Fatal server error:
[  1026.254] (EE) no screens found(EE)
[  1026.254] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[  1026.255] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[  1026.255] (EE)
[  1026.255] (EE) Server terminated with error (1). Closing log file.`

McPrince96, 2020-02-09:
[  1026.249]
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[  1026.249] Build Operating System: Linux 4.15.0-48-generic armv8l Raspbian
[  1026.250] Current Operating System: Linux octopi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l
[  1026.250] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=533 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27$
[  1026.250] Build Date: 05 June 2019  12:49:54PM
[  1026.250] xorg-server 2:1.20.4-1+rpt1 (https://www.debian.org/support)
[  1026.250] Current version of pixman: 0.36.0
[  1026.250]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[  1026.250] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[  1026.251] (==) Log file: "/var/log/Xorg.0.log", Time: Sun Feb  9 15:12:24 2020
[  1026.252] (==) Using config file: "/etc/X11/xorg.conf"
[  1026.252] (==) Using config directory: "/etc/X11/xorg.conf.d"
[  1026.252] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[  1026.254] Parse error on line 8 of section InputClass in file /etc/X11/xorg.conf.d/99-calibration.conf
        The Section keyword requires a quoted string to follow it.
[  1026.254] (EE) Problem parsing the config file
[  1026.254] (EE) Error parsing the config file
[  1026.254] (EE)
Fatal server error:
[  1026.254] (EE) no screens found(EE)
[  1026.254] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[  1026.255] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[  1026.255] (EE)
[  1026.255] (EE) Server terminated with error (1). Closing log file.

hunagyp, 2020-02-09:
@McPrince96 I think you made typo in your config file, fix it.

[ 1026.254] Parse error on line 8 of section InputClass in file /etc/X11/xorg.conf.d/99-calibration.conf
The Section keyword requires a quoted string to follow it.

JeffB42, 2020-12-04:
@McPrince96 Which screen are you using?  If you are using a 3.5" 480x320 TFT screen, could you follow the steps listed in the wiki (located at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) and then post a reply to this issue whether the steps listed work and the issue is resolved?  I'd like to start closing out all the display-related issues.

If it doesn't work,
1) make sure you use a Raspberry Pi 3B or Raspberry Pi 4
2) start with a fresh install of OctoPi
3) please take a picture of the screen

Thanks, 

Jeff

JeffB42, 2021-01-06:
@McPrince96 Is this still an issue?  Did you get your display working?

closed 2021-03-27 by JeffB42