Z-Bolt/OctoScreen #90: Rotate 90degree for portrait mode

iosifLuchian, 2020-02-11:
is there a possibility to rotate 90 degree to use the app in the portrait mode?
i tried in the /boot/config.txt  the line display_rotate =1 but the app not work anymore only if i comment the line 
if there is a solution please let me know
hank you

mash55, 2020-03-09:
is your screen configured for portrait?

goeland86, 2020-03-15:
I'm actually interested in this as well.

The Manga Screen 2 is a 6" panel for smartphones, full HD, but rotated 90 degrees... It's frustrating to not be able to use it properly.

Woodpeckercz, 2020-05-03:
I'm insterested in this as well, I have original raspberry pi touchscreen and managed to rotate it 90deg, but octoscreen doesn't load after I did that.

JeffB42, 2021-03-27:
@iosifLuchian sorry, but it doesn't. OctoScreen is set up into a 4 column x 3 row display layout. It's written using GTK, which isn't a responsive design UI library (meaning, it doesn't automatically adjust its layout depending on the screen orientation). This layout is pretty entranced in OctoScreen, and at this time I don't plan to add support for a portrait layout.

closed 2021-03-27 by JeffB42

goeland86, 2021-08-05:
@JeffB42 my issue is that the MangaScreen 2 is expecting a signal in portrait full HD (i.e. 1080x1920), and I can't figure out how to get the armbian-based image I use to rotate the display using Xorg conf alone... Beaglebone is a bit of a PITA with that. I should've been a bit clearer in my explanation perhaps. As it is I get the display essentially shrunk to a 1080 wide screen and have to use it in the top third of the screen rotated vertically 🤷