Z-Bolt/OctoScreen #91: No ability to adjust z offset during print

dgwood12, 2020-02-11:
For users with an ABL tool such as the BL touch or EZABL, it's important to dial in the Z offset to get good adhesion and great prints. In Marlin, one of the nicest things is the ability to adjust the z offset on the fly during a print. This allows users to finely tune the z offset during the first layer for perfect results. It would be extremely helpful to add the ability to adjust the z offset during a print which currently is not possible with OctoScreen. Thanks!

jalfeld, 2020-04-20:
Would love to see this also, this is critical if octoscreen is going to be able to replace the standard lcd on printers.

elictes, 2020-08-31:
This would be awesome.

JeffB42, 2020-09-08:
I'll add this to the feature request backlog.

werddomain, 2021-01-29:
I will switch to OctoPrint when this feature is avaliable. Currently i use OctoDash.