Z-Bolt/OctoScreen #92: How to install a custom command button

BelSD, 2020-02-12:
I have custom commands in Octoprint, but I don't see them in Octoscreen ...
[image]
[image]
[image]
[image]

maciekish, 2020-04-15:
Same issue here, any news please?

mtbrown4ayl, 2020-05-23:
Same here. What is the process? Are we missing something???

JeffB42, 2021-03-27:
@BelSD version 27.0 was recently released, and fixed some bugs pertaining to custom commands and custom buttons.

closed 2021-03-27 by JeffB42