Z-Bolt/OctoScreen #93: Prusa MMU2 filament Select

NightSkySK, 2020-02-14:
Is it possible to implement fillament selection on OctoScreen base on mmu2filamentselect (https://github.com/derPicknicker1/OctoPrint-Mmu2filamentselect) plugin or as independent feature?

NightSkySK, 2021-02-19:
Recently the repository of plugin was moved to @troecker https://github.com/tkoecker/OctoPrint-Mmu2filamentselect

JeffB42, 2021-02-21:
@NightSkySK I don't own a MMU and haven't looked at this plug-in yet.  For the upcoming 2.7 release, I've added support for single nozzle/multiple extruder printers, like the MMU.  In the Filament panel, I've added a extruder select button.  If the printer profile that's defined in OctoPrint has multiple extruders but also a shared nozzle, this button appears and allows you to select which extruder the load/unload and extrude/retract commands are issued for.

[image]

Is this what you are looking for, or is it something else?

NightSkySK, 2021-02-22:
@JeffB42 Prusa has implemented [custom Gcode commands](https://help.prusa3d.com/en/article/prusa-specific-g-codes_112173) for MMU unit:

> T-codes
T<extruder nr.> - select extruder in case of multi extruder printer. select filament in case of MMU_V2.

>For MMU_V2:
T<n> Gcode to extrude at least 38.10 mm at feedrate 19.02 mm/s must follow immediately to load to extruder wheels.T? Gcode to extrude shouldn't have to follow, load to extruder wheels is done automatically Tx Same as T?, except nozzle doesn't have to be preheated. Tc must be placed after extruder nozzle is preheated to finish filament load. Tc Load to nozzle after filament was prepared by Tc and extruder nozzle is already heated.

MMU2fillamentselect plugin is detecting `Tx` command in Gcode and display at the Octoprint page the filament select window.

[image]

If the filament is not selected at Octoprint (after set time e.g. 30 sec) the Filament select option is displayed at printer display and can be selected by printer knob. 

Example Gcode for MMU printer in Single color mode looks like this:

```;TYPE:Custom
M862.3 P "MK3SMMU2S" ; printer model check
M862.1 P0.4 ; nozzle diameter check
M115 U3.9.3 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S240 ; set extruder temp
M140 S85 ; set bed temp
Tx
M190 S85 ; wait for bed temp
M109 S240 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling

;go outside print area
G1 Y-3.0 F1000.0
G1 Z0.4 F1000.0
; select extruder
Tc
```

JeffB42, 2021-02-27:
I haven't looked at that plugin or its implementation yet.  I was wondering though, in the screenshot you provided, there are five buttons.  Do each send (essentially) static g-code?  If so, why not create custom controls in OctoPrint (https://docs.octoprint.org/en/master/features/custom_controls.html) either defining them manually in OctoPrint's config.yaml file, or visually using the "Custom Control Editor" plug-in?  (I recommend using the plug-in).

OctoPrint looks for custom controls, and then displays each one that's found as a button in the Controls panel:
[image]
(this images doesn't list any, but this is where they would be and there would be one of these https://github.com/Z-Bolt/OctoScreen/blob/master/styles/z-bolt/images/custom-script.svg for every custom control that's defined in OctoPrint)

So, TL;DR... would creating custom controls fulfill this FR?

I'm not opposed to making a custom panel just for the Prusa MMU, but I am trying to keep OctoScreen printer-agnostic, and if there exists an existing solution, I'd prefer to go with that.  Not owning an MMU, I can't tell if those buttons send (essentially) static g-code, or if there's more involved (and if there's more involved, then yea, a new Panel and new buttons in OctoScreen will need to be made).

NightSkySK, 2021-03-09:
@JeffB42 The idea to use the existing functionality of Octoprint and OctoScreen is great, but I don't fully understand if that can be triggered by some g-code sequence? The buttons which are shown above as filament select window, are triggered in case if 'Tx' g-code command is detected during the print...

JeffB42, 2021-03-27:
v2.7.0 was just released.

JeffB42, 2021-03-28:
Update: I released 2.7.0 today, but https://github.com/Z-Bolt/OctoScreen/issues/274 was reported.  I plan to put out a dot release next weekend (2.7.1) to address this bug, and any others which might be reported this week.