Z-Bolt/OctoScreen #94: Updated install steps, please

Stady234, 2020-02-23:
I am using a fresh install image of Octopi with a 3.5" touch screen...
 ```
OctoPrint version : 1.3.12
OctoPi version    : 0.17.0
```
I followed the "Install - Dependencies" section.. which errors out saying multiple packages are required but can not install... and it recommends to use "apt --fix-broken install" .. then I re-ran the dependencies script line... no more errors...

I then installed the ".deb" package... there are a few errors, but i do not see how to fix...

I edit the "basic Configuration" file.. adding the API key and adjusting screen size per notes..

The HDMI output shows an Octopi Logo (on the lower right of screen) saying it can not initialize printer.  The 3.5 Screen stays on the Octopi login...

What am I missing?  I had tried the "docker part" but it states there was an error and it is suspending input.. so i can not get the the ">make build" section of the guide...

eggerc, 2020-02-24:
I had the same issue showing Octopi login. I tried to reinstall the deb and it threw errors. Checking those errors showed "No screens found". Using the link in the octoscreen installation guide fixed it.

Stady234, 2020-02-24:
I am not getting the "No screens found"... i tried to re-install the "no auto login" version of the install and the 3.5" still only stays on the Login screen.  the HDMI output is looking better now that I adjusted the screen resolution.  I get what looks like the correct start screen, but nothing on the 3.5 TFT....

Stady234, 2020-02-27:
So it looks like the install instructions ASSUME the end user installed the octoprint desktop feature on the PI... now that that is installed i have an active screen with like pictures from install.  Once i get repeatable install steps i will post.

komandrik, 2020-02-27:
@Stady234 я имею эту проблему. Экран DSI. После включения принтера и загрузки я остаюсь на экране консоли. Когда в консоли я запускаю команду 
sudo systemctl restart octoscreen.service

JeffB42, 2021-01-05:
@Stady234 there are some updated install steps:
https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display
https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

Please post a reply and let me know if this is still an issue.

JeffB42, 2021-02-04:
Closed due to inactivity.

closed 2021-02-04 by JeffB42