Z-Bolt/OctoScreen #95: Errors on install...

Stady234, 2020-02-27:
pi@octopi:~ $ sudo dpkg -i octoscreen_2.5-1_armhf.deb
Selecting previously unselected package octoscreen.
(Reading database ... 87773 files and directories currently installed.)
Preparing to unpack octoscreen_2.5-1_armhf.deb ...
Unpacking octoscreen (2.5-1) ...
Setting up octoscreen (2.5-1) ...
Installing service /lib/systemd/system/lightdm.service
Failed to preset unit: File /etc/systemd/system/display-manager.service already exists and is a symlink to /lib/systemd/system/lightdm.service.
/usr/bin/deb-systemd-helper: error: systemctl preset failed on octoscreen.service: No such file or directory
Job for octoscreen.service failed because the control process exited with error code.
See "systemctl status octoscreen.service" and "journalctl -xe" for details.
pi@octopi:~ $

AlbanT, 2020-02-29:
I have the same issue :(

> pi@octopi:~ $ sudo dpkg -i octoscreen_2.5-1_armhf.deb
> (Reading database ... 55405 files and directories currently installed.)
> Preparing to unpack octoscreen_2.5-1_armhf.deb ...
> Unpacking octoscreen (2.5-1) over (2.5-1) ...
> Setting up octoscreen (2.5-1) ...
> Job for octoscreen.service failed because the control process exited with error code.
> See "systemctl status octoscreen.service" and "journalctl -xe" for details.

systemctl status octoscreen.service gives me:

> pi@octopi:~ $ systemctl status octoscreen.service
> ● octoscreen.service - OctoScreen
>    Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
>    Active: activating (start-post) since Sat 2020-02-29 22:15:34 GMT; 10s ago
>  Main PID: 6389 (xinit); Control PID: 6410 (bash)
>     Tasks: 21 (limit: 2077)
>    Memory: 24.0M
>    CGroup: /system.slice/octoscreen.service
>            ├─6389 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor
>            ├─6390 /usr/lib/xorg/Xorg :0 -nolisten tcp -nocursor
>            ├─6398 /usr/bin/OctoScreen
>            ├─6410 /bin/bash /etc/octoscreen/disablescreenblank.sh 0
>            └─6411 sleep 10
> 
> Feb 29 22:15:34 octopi xinit[6389]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
> Feb 29 22:15:35 octopi xinit[6389]: time="2020-02-29T22:15:35Z" level=info msg="OctoPrint's config file found: \"/home/pi/.octoprint/config.yaml\"" file=main.go func=main.readConfig line=102
> Feb 29 22:15:35 octopi xinit[6389]: time="2020-02-29T22:15:35Z" level=info msg="Using \"http://localhost:5000\" as server address" file=main.go func=main.init.0 line=52
> Feb 29 22:15:35 octopi xinit[6389]: time="2020-02-29T22:15:35Z" level=info msg="Found API key at \"/home/pi/.octoprint/config.yaml\" file" file=main.go func=main.init.0 line=58
> Feb 29 22:15:36 octopi xinit[6389]: time="2020-02-29T22:15:36Z" level=debug msg="New background task started" file=common.go func="ui.(*BackgroundTask).Start" line=136
> Feb 29 22:15:36 octopi xinit[6389]: time="2020-02-29T22:15:36Z" level=error msg="unexpected status code: 403" file=ui.go func="ui.(*UI).loadSettings" line=194
> Feb 29 22:15:38 octopi xinit[6389]: time="2020-02-29T22:15:38Z" level=error msg="unexpected status code: 403" file=ui.go func="ui.(*UI).loadSettings" line=194
> Feb 29 22:15:40 octopi xinit[6389]: time="2020-02-29T22:15:40Z" level=error msg="unexpected status code: 403" file=ui.go func="ui.(*UI).loadSettings" line=194
> Feb 29 22:15:42 octopi xinit[6389]: time="2020-02-29T22:15:42Z" level=error msg="unexpected status code: 403" file=ui.go func="ui.(*UI).loadSettings" line=194
> Feb 29 22:15:44 octopi xinit[6389]: time="2020-02-29T22:15:44Z" level=error msg="unexpected status code: 403" file=ui.go func="ui.(*UI).loadSettings" line=194

And journalctl -xe reports:

> pi@octopi:~ $ journalctl -xe
> -
> - An ExecStart= process belonging to unit octoscreen.service has exited.
> -
> - The process' exit code is 'exited' and its exit status is 1.
> Feb 29 22:17:58 octopi systemd[1]: octoscreen.service: Failed with result 'exit-code'.
> - Subject: Unit failed
> - Defined-By: systemd
> - Support: https://www.debian.org/support
> -
> - The unit octoscreen.service has entered the 'failed' state with result 'exit-code'.
> Feb 29 22:17:58 octopi systemd[1]: Failed to start OctoScreen.
> - Subject: A start job for unit octoscreen.service has failed
> - Defined-By: systemd
> - Support: https://www.debian.org/support
> -
> - A start job for unit octoscreen.service has finished with a failure.
> -
> - The job identifier is 3006 and the job result is failed.
> Feb 29 22:17:58 octopi systemd[1]: octoscreen.service: Service RestartSec=100ms expired, scheduling restart.
> Feb 29 22:17:58 octopi systemd[1]: octoscreen.service: Scheduled restart job, restart counter is at 23.
> - Subject: Automatic restarting of a unit has been scheduled
> - Defined-By: systemd
> - Support: https://www.debian.org/support
> -
> - Automatic restarting of the unit octoscreen.service has been scheduled, as the result for
> - the configured Restart= setting for the unit.
> Feb 29 22:17:58 octopi systemd[1]: Stopped OctoScreen.
> - Subject: A stop job for unit octoscreen.service has finished
> - Defined-By: systemd
> - Support: https://www.debian.org/support
[... 15 lines cut ...]
> Feb 29 22:17:58 octopi xinit[7290]: Current Operating System: Linux octopi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l
> Feb 29 22:17:58 octopi xinit[7290]: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000
> Feb 29 22:17:58 octopi xinit[7290]: Build Date: 05 June 2019  12:49:54PM
> Feb 29 22:17:58 octopi xinit[7290]: xorg-server 2:1.20.4-1+rpt1 (https://www.debian.org/support)
> Feb 29 22:17:58 octopi xinit[7290]: Current version of pixman: 0.36.0
> Feb 29 22:17:58 octopi xinit[7290]:         Before reporting problems, check http://wiki.x.org
> Feb 29 22:17:58 octopi xinit[7290]:         to make sure that you have the latest version.
> Feb 29 22:17:58 octopi xinit[7290]: Markers: (--) probed, (**) from config file, (==) default setting,
> Feb 29 22:17:58 octopi xinit[7290]:         (++) from command line, (!!) notice, (II) informational,
> Feb 29 22:17:58 octopi xinit[7290]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[... cut ...]
> Feb 29 22:18:04 octopi xinit[7290]: time="2020-02-29T22:18:04Z" level=error msg="unexpected status code: 403" file=ui.go func="ui.(*UI).loadSettings" line=194
> Feb 29 22:18:06 octopi xinit[7290]: time="2020-02-29T22:18:06Z" level=error msg="unexpected status code: 403" file=ui.go func="ui.(*UI).loadSettings" line=194
> Feb 29 22:18:08 octopi xinit[7290]: time="2020-02-29T22:18:08Z" level=error msg="unexpected status code: 403" file=ui.go func="ui.(*UI).loadSettings" line=194
> lines 2073-2138/2138 (END)
> 

Not sure what to make of this :)

AlbanT, 2020-02-29:
Not sure what fixed it but for me it works now..

OutsourcedGuru, 2020-03-11:
I will note that I'm talking about [this on the OctoPrint forum](https://community.octoprint.org/t/octoscreen-installation-schlagt-fehl/17033?u=outsourcedguru). I'm suggesting that the OctoScreen author assumes that the OctoPrint daemon is started from `/etc/systemd/system` rather than from `/etc/init.d`. The OctoPi IMG comes with the latter, btw.

mannyr7, 2020-06-18:
So what is the fix to correct? Create symlink from OctoPrint daemon at /etc/init.d to /etc/systemd/system?

scifiaholic, 2020-08-05:
Hey so I want to echo that the .deb package still makes this assumption that the OctoPrint daemon is started from /etc/systemd/system rather than from /etc/init.d I'm too much of a novice to know how to fix this... can someone please help?

bepstein111, 2020-08-19:
> Preparing to unpack .../octoscreen_2.5-1_amd64.deb ...
> Unpacking octoscreen (2.5-1) over (2.5-1) ...
> Setting up octoscreen (2.5-1) ...
> Installing service /lib/systemd/system/gdm3.service
> Failed to preset unit: File /etc/systemd/system/display-manager.service already exists and is a symlink to /lib/systemd/system/gdm3.service.
> /usr/bin/deb-systemd-helper: error: systemctl preset failed on octoscreen.service: No such file or directory

So I'm installing this on an actual Ubuntu Server installation, so I am dealing with systemd. I've gotten octoscreen running by running `service gdm stop` *insert sudo password* `service octoprint start` *insert sudo password* , but setting up and autostart file to run those lines seems too hack-y for my liking. i tried `sudo rm /etc/systemd/system/display-manager.service` and then running `dpkg -i` right after, and still, same thing. Any insights would be super appreciated.

JeffB42, 2020-08-19:
@bepstein111 - I don't know what the issue is but here are my suggestions:
1) instead of installing on an actual Ubuntu Server, install OctoPi (either install using a disk imager and wipe your current memory card, or get another memory card and install via disk image onto it)
2) If you are running Ubuntu, does that mean you have two displays connected?  (perhaps a large monitor to display Ubuntu, and a TFT or HyperPixel or some other small device to display OctoScreen?)  If so, try installing and configuring using only the display... this is just a guess, but maybe having a GUI running is interfering with OctoScreen running.  I know OctoScreen uses GTK, but I have no idea it if's incompatible with also running Ubuntu, or one of the other GUI systems for Linux.

bepstein111, 2020-08-19:
> @bepstein111 - I don't know what the issue is but here are my suggestions:
>
> 1. instead of installing on an actual Ubuntu Server, install OctoPi (either install using a disk imager and wipe your [...]

Right so I should have specified, I’m running on a fairly decently powered ryzen setup (some old parts I had lying around, to improve performance), so no Pi here. As for other monitors, I have an old GeForce 8400GS and used a VGA monitor for installation but then unplugged it. Now I’m using a DVI to hdmi adapter to plug into the touch screen. Possibly it “remembered” the vga settings and is trying to do something with that?

It seems to me the root cause is something to do with the process involving replacing the current desktop manager (gdm or lightdm, whichever) with octoscreen. Ideally, there wouldn’t have been any desktop manager setup at all previously, but it seems like xorg (maybe something related) sets it up on installation.

closed 2020-08-20 by Stady234