Z-Bolt/OctoScreen #96: Screeensaver?

mjgtp, 2020-02-27:
Been using OctoScreen for a short time, but noticed the screen never sleeps.  Is this something configurable or an oversight?  Some of these prints can take days and having the screen on with the icons in the same place could result in burn-in.  Thoughts?

bolsoncerrado, 2020-03-06:
Im also curious!

OutspokenPlanet, 2020-03-18:
Me three.

tekmad, 2020-04-15:
Screen sleep would be nice.

mrkag, 2020-05-23:
I would love to see a dimly lit Octoprint mascot (does it have a name?) bouncing around the screen when the printer isn't in use.

pitchuk, 2020-05-27:
Me too, been trying to find a way to do this for ages

daimon82, 2020-05-27:
I managed to implement simple screen saver by editing the xorg config:
sudo nano /etc/X11/xorg.conf

and paste the following code:

Section "ServerFlags"
Option "blank time" "5"
Option "standby time" "10"
Option "suspend time" "15"
Option "off time" "20"
EndSection

ChaosBlades, 2020-05-27:
Pretty sure this project is dead. No commits since December of last year. OctoDash has had a screensaver for quite a while and is in active development.

https://github.com/UnchartedBull/OctoDash

JeffB42, 2020-07-31:
This issue is a duplicate of https://github.com/Z-Bolt/OctoScreen/issues/132

closed 2020-07-31 by JeffB42