Z-Bolt/OctoScreen #98: Is it possible to use a mouse?

clintonblackmore, 2020-03-04:
Thank you very much for building this.

I realize the goal is to work with touch screens, but I've installed this software on an old netbook; (my R-Pi 1Bs feel too slow for OctoPi). It has a built-in trackpad, and I could plug in a mouse, but it doesn't have a touch screen.

I was wondering if there are any configuration options to enable mouse/trackpad support, or, indeed, if your underlying GUI library supports them.

(Yes, I could try out the TouchUI interface, but my netbook is sluggish when running a modern browser).

Cheers.

kabrokes, 2020-04-23:
I'm with you, I have a screen that is supposed to be touch screen, but I have a keyboard and mouse attached.  I would like to use a mouse as well.

JeffB42, 2020-07-29:
@clintonblackmore and @kabrokes - it is possible.  I think you need to add this:
xserver-command=X -nocursor
...probably to your bash file, or something that runs automatically at startup.

See https://raspberrypi.stackexchange.com/questions/53127/how-to-permanently-hide-mouse-pointer-or-cursor-on-raspberry-pi

closed 2021-02-04 by JeffB42