Z-Bolt/OctoScreen #100: Unload filament not working

daimon82, 2020-03-06:
I have prusa mk3s and raspberry pi 4 with octoprint 1.4.0, load filament works correctly, but unload filament only starts motors but don't do anything.

SW0rden, 2020-03-11:
Same issue for my CR-20 and RPi 3.  Warm up nozzle and then touch "Unload" and nothing happens.
Version 1.4.0 (0.1)

rglory, 2020-04-25:
Actually neither of them works properly. The problem is values for load/unload filament are hardcoded huge and unrealistic - 800mm to unload and 600mm to load. Both are blocked by firmware as too large. Loading seems to work as there is additional 120mm load in slower speed (in addition to 600) which accepted by firmware.

JeffB42, 2021-01-05:
@rglory and @Protean-Man I'm going to be fixing this in 2.7.  Do either of you have have preference as to what the default load and unload lengths should be?  I agree, 800mm and 600mm seem too long

Protean-Man, 2021-01-05:
@JeffB42 for me they are actually too short. I have a load/unload length of 1640 on a new printer prototype.
I don't think any default will be adequate for such a varied length in lines among the printers out there. Direct drives have a load length of around 100 and bowdens can be very long. My prototype is currently a dual drive with a load of 1640 so as long as it is editable then you can put whatever you would like.

Protean-Man, 2021-01-05:
let's say 500 as a decent middle ground for most users?

JeffB42, 2021-03-27:
Version 2.7.0 was just released.

closed 2021-03-27 by JeffB42